noble/noble #670: No compatible USB Bluetooth 4.0 device found!

kingio, 2017-08-28:
**O.S:** Windows 10 32b
**Arch**: ia32
**USB id**: 0x8087, 0x0a2a

I've did the following install steps:
1. Admin cmd -> `npm install --global --production windows-build-tools` ->Success
2. Zadig -> Select Intel Wireless Bluetooth -> Install Driver -> Success
3. Download repo -> `npm install` -> run an example -> Error `No compatible USB Bluetooth 4.0 device found!`

What did i do wrong?

kingio, 2017-08-28:
I've checked the code and while:
https://github.com/sandeepmistry/node-bluetooth-hci-socket/blob/master/lib/usb.js#L66
does contain my USB ID, the latest release here doesn't:
https://github.com/sandeepmistry/node-bluetooth-hci-socket/releases

Master:
```js
usb.findByIds(0x0CF3,0xE300) || // Qualcomm Atheros QCA61x4 
      usb.findByIds(0x0a5c, 0x21e8) || // Broadcom BCM20702A0
      usb.findByIds(0x19ff, 0x0239) || // Broadcom BCM20702A0
      usb.findByIds(0x0a12, 0x0001) || // CSR
      usb.findByIds(0x0b05, 0x17cb) || // ASUS BT400
      usb.findByIds(0x8087, 0x07da) || // Intel 6235
      usb.findByIds(0x8087, 0x07dc) || // Intel 7260
      usb.findByIds(0x8087, 0x0a2a) || // Intel 7265
      usb.findByIds(0x0489, 0xe07a) || // Broadcom BCM20702A1
      usb.findByIds(0x0a5c, 0x6412) || // Broadcom BCM2045A0
      usb.findByIds(0x050D, 0x065A); // Belkin BCM20702A0
````
Release:
```
 usb.findByIds(0x0a5c, 0x21e8) || 
usb.findByIds(0x19ff, 0x0239) || 
usb.findByIds(0x0a12, 0x0001) || 
usb.findByIds(0x0b05, 0x17cb) || 
usb.findByIds(0x8087, 0x07dc) || 
usb.findByIds(0x0489, 0xe07a);
````

sandeepmistry, 2017-09-05:
@kingio yep, a new `bluetooth-hci-socket` release is needed.