noble/noble #672: a more descriptive documentation

tsadak, 2017-09-03:
Hello

Noble is doubtless a very good product and thanks to the authors and backers and sponsors. 
But being a newbie to the beacon domain i cannot easily figure out how to use it. the documentation describes the function syntax only. But i don't exactly know what function to use for what purpose. can anybody help us who feel the same about that?

Thanks in advance

JDrakeC, 2018-02-02:
Any luck?  I'm having the same issue.  instructions are disjointed ive tried to npm install everything 6 ways from Sunday and i still cant get example.js working with node...

jacobrosenthal, 2018-02-02:
Not sure what example.js is. A tutorial for you
```
$ git clone git@github.com:sandeepmistry/noble.git
$ cd noble
$ npm i
$ node examples/advertisement-discovery.js 
peripheral discovered (bd384f06c8824e88ae1ebe157ea495d1 with address <d0:03:4b:31:75:f2, unknown>, connectable true, RSSI -91:
	hello my local name is:
		Apple TV
	can I interest you in any of the following advertised services:
		[]
	here is my manufacturer data:
		"4c0010050110686541"

peripheral discovered (e91661d3b859460ba84046e3c779e64c with address <unknown, unknown>, connectable true, RSSI -95:
	hello my local name is:
		undefined
	can I interest you in any of the following advertised services:
		[]
	here is my manufacturer data:
		"4c0010050b10745f4b"

^C
$ node examples/peripheral-explorer.js d0:03:4b:31:75:f2
peripheral with ID bd384f06c8824e88ae1ebe157ea495d1 found
  Local Name        = Apple TV
  Manufacturer Data = 4c0010050110686541
  Service Data      = []
  Service UUIDs     = 

services and characteristics:
$ 
```