noble/noble #673: noble warning: unknown peripheral xxxxxxxxxxxxx

ljonget, 2017-09-05:
Hi,

I'm trying to connect to a BLE sensor (xiaomi "mi plant") and when using noble's method "peripheral.readHandle" after connected to it, I have a warning : "noble warning: unknown peripheral xxxxxxxxxxxxx".

how to dig into it and see what's wrong 

thanks

PS : noble is working on another BLE sensor "Parrot Flower power"

ljonget, 2017-09-05:
here is the code used : 
```javascript
var noble = require('../index');

noble.on('stateChange', function(state) {
  if (state === 'poweredOn') {
    noble.startScanning();
  } else {
    noble.stopScanning();
  }
});

var peripheralIdOrAddress = "XX:XX:XX:XX:XX:XX"

noble.on('scanStop', function(peripheral) {
  process.exit(0)
});
/*
noble.on('discover', function(peripheral) {
  //peripheral.once('connect', function(){   }  );
    console.log ("found : id "+peripheral.id+" addr "+peripheral.address)
    peripheral.connect();
  }
});
*/
noble.on('discover', function(peripheral) {
  if (peripheral.id === peripheralIdOrAddress || peripheral.address === peripheralIdOrAddress ){
    peripheral.connect(function(error) {
     console.log('connected to peripheral: ' + peripheral.uuid);
            peripheral.readHandle(0x0038, function(error, data){
       console.log(data);
       noble.stopScanning();
       }
      );
    });
  }
});
```

londonoliver, 2017-09-06:
What operating system are you on?

If you are running on Linux, you could run HCI dump capture: `sudo hcidump -t -x`

If you are on Mac, you can try [Packet Logger](https://developer.apple.com/library/content/documentation/DeviceDrivers/Conceptual/Bluetooth/BT_Bluetooth_On_MOSX/BT_Bluetooth_On_MOSX.html#//apple_ref/doc/uid/TP30000997-CH215-TPXREF10) (search "Packet Logger")

Also, you can run debug logs: `sudo DEBUG=* node <file>.js`

ljonget, 2017-09-07:
I'm running a Debian 8.7 on a Synology NAS 
here is the hcidump and the DEBUG output
[debug_output.txt](https://github.com/sandeepmistry/noble/files/1285827/debug_output.txt)
[hcidump.txt](https://github.com/sandeepmistry/noble/files/1285828/hcidump.txt)

bepitulaz, 2017-10-10:
I got the same issue with Xiaomi Flora. I'm using Raspbian in Raspberry Pi 3 and the Xiaomi Flora's firmware version 3.1.8.

@ljonget have you solved your problem?

ljonget, 2017-10-26:
@bepitulaz : I couldn't solved the problem, I switched to python 

but I'm still interested by the solution to avoid having nodejs and python running

GSanCap, 2018-03-05:
Any news!? i have the same problem with the Xiaomi Mijia BLE Hygrometer, i try to write an Homebridge PlugIn, help accepted ;)

Regards