noble/noble #679: No event 'stateChange' on macOS High Sierra

piotrdubiel, 2017-09-19:
*Setup*
noble: 1.8.1
node: 7.7.2
macOS High Sierra 10.13 GM (17A362a)

*Description*
Noble never emits 'stateChange' events on High Sierra, therefore scanning cannot be started.

*Code*
```
var noble = require('noble');

noble.on('stateChange', function(state) {
  console.log(state); // never called
});
```

zwaldowski, 2017-09-20:
High Sierra has replaced `blued` with `bluetoothd`, and with it a replacement to the previous XPC messages.

mbruschi, 2017-09-20:
@zwaldowski do you have any idea what we have to do in order to make it work again? How can I help?

Axadiw, 2017-09-22:
We're working on it here: https://github.com/PolideaInternal/noble/tree/macos_highsierra

Unfortunately https://github.com/sandeepmistry/osx-ble-peripheral isn't working anymore on HighSierra (we're not able to swizzle `CBXpcConnection` class there).

We've been sniffing XPC traffic using http://newosxbook.com/tools/XPoCe.html  and decompiling `bluetoothd` using https://www.hopperapp.com  and we've managed to retrieve some of the correct ID's that are being sent byt XPC to `bluetoothd` on HighSierra.

If someone could test it, and fix all not working calls, we'll be very appreciated.

Generally I think that this XPC method should be abanddoned in `noble` and some legit implementation basing on `CoreBluetooth` should be used there (I think that https://github.com/sandeepmistry/noble/tree/more-test is a good start for it)

jacobrosenthal, 2017-12-26:
I have a hopefully complete high sierra PR in 
https://github.com/sandeepmistry/noble/pull/727

Needs testing though.

You can have npm install from my git branch with 
`npm install git://github.com/jacobrosenthal/noble.git#highsierra` or edit your package.json and add the branch to your dependencies
```
  "dependencies": {
    "noble": "git://github.com/jacobrosenthal/noble.git#highsierra"
  }
```

Ezor, 2017-12-29:
@jacobrosenthal Just tested it on a little project and it works like a charm

klategan, 2017-12-29:
@jacobrosenthal Working perfectly. Thanks!

stefanclayster, 2018-01-04:
Ive tried it with muse-lsl (https://github.com/urish/muse-lsl) and it connects to the device but wont get any data from the characteristics. Has anything changed in its interface?

jacobrosenthal, 2018-01-04:
@stefanclayster can you enable debug logging for the bindings layer and report back
`DEBUG=highsierra-bindings node yourscripthere.js`

jacobrosenthal, 2018-01-08:
ping @stefanclayster you're the only one to report in with problems. Were you able to solve those? If not can you enable debugging by running like `DEBUG=highsierra-bindings node yourscripthere.js`

stefanclayster, 2018-01-08:
Sorry I haven't responded yet, I am on a skiing holiday.I did run with the debug flag and i got too much data outputted to know where the bug is/what is relevant. I will have to investigate further.

jacobrosenthal, 2018-01-08:
Please send that log over, well look at it.

stefanclayster, 2018-01-15:
[debuglog.txt](https://github.com/sandeepmistry/noble/files/1631150/debuglog.txt)
Here is the log.

stefanclayster, 2018-01-15:
I reckon that log was too short. Longer version below with data from characteristics.

[debuglog2.txt](https://github.com/sandeepmistry/noble/files/1632155/debuglog2.txt)

jacobrosenthal, 2018-01-15:
@stefanclayster Seems like all xpcids are being caught and you're getting a ton of reads (83). Make sure you pull fresh? Its had some changes, but otherwise I think youll need to poke around and debug in this function to see why that function or the emit inside it isnt being called for you https://github.com/sandeepmistry/noble/pull/727/files#diff-81c793ee25a4a5820d33c0d4c9beeea0R519

To explain the log:
48s peripheral founds
45 stop scanning
57 response to stop scanning we dont do anything with 
46 49 connect/connected
62 72 discover services / did discoverservices
75 discover characteristics
60 params update we dont do anything with
77 response discover characteristics
81 notify / 86 response notify on a bunch of chars
79 write value a couple times
83 response read value over and over again

stefanclayster, 2018-01-15:
Cool! Ill dig into it! Thank you.

stefanclayster, 2018-01-17:
Is the emitter on 'characteristicvaluechanged' removed? I cannot find it in the repo at all (github search) while it exists in noble repo.

jacobrosenthal, 2018-01-17:
That appears to be a web bluetooth specific event thats consumed by the web bluetooth bindings layer. Shouldn't have anything to do with macosx support. 

High Sierra consumes the kCBMsgId83 event for reads and subscribe notifications triggering 'read' emits to characteristic.js which are consumed in noble.js and trigger  'data' emits for user facing code.

weisjohn, 2018-01-26:
Switching to @jacobrosenthal's fork allowed me to get events right away. Thanks.

closed 2018-01-31 by sandeepmistry

ainthek, 2018-12-19:
Do we need another patch for Mohave now ?