noble/noble #680: Noble state problems on MacOS High Sierra

alexander-petrov-wacom, 2017-09-20:
We are facing issues related to the internal state of the library (regarding the state of the Bluetooth module). There are two parts of the problem:

1. As stated in issue #679 - `stateChanged` event  is never emitted.
2. Polling the state from the library, using `noble.state` is also not working. It returns `undefined`.

Therefor, regarding these two issues (I am sure that they have the same origin) - none of the other functionalities of noble are working properly on MacOS High Sierra.

I would please ask you to help us fix this issue because very soon this will impact our clients and render us unable to provide them with the service they expect.

Our setup:
Node.js - v7.10.1
Electron - v1.7.6
MacOS - v10.13 Beta (High Sierra)

mbruschi, 2017-09-20:
@alexander-petrov-wacom #679 description has been updated.

jacobrosenthal, 2017-12-23:
Duplicate of #679

closed 2017-12-23 by jacobrosenthal