noble/noble #681: Can't detect Nexus 5

lastmjs, 2017-09-21:
So, I've started a scan:

```
const noble = require('noble');

noble.on('discover', (peripheral) => {
        console.log(peripheral);
        //updateRssiLoop(peripheral);
});

noble.on('stateChange', (state) => {
        console.log(state);

        if (state === 'poweredOn') {
                noble.startScanning([], true);
        }
});
```

I've turned on Bluetooth on my Nexus 5 device, which has Bluetooth 4.0, and I don't seem to be able to discover the peripheral. Is there anything basic that I'm missing?

lastmjs, 2017-09-21:
Are BLE advertisements required on a device to be able to discover it? Apparently the Nexus 5 does not support advertisements: http://appinventor.pevest.com/?p=1415

Is this the root of the issue?

elovin, 2017-10-27:
My nexus 5 running LineageOS (Android 7.1.2) does support Advertising, maybe you are running an old Android version ?

monteslu, 2017-10-28:
Are you running your noble example from a desktop attempting to connect to a service on your nexus 5?  Any particular service?

You can have your phone advertise a service using something like the BLE simulator app:  https://play.google.com/store/apps/details?id=io.github.webbluetoothcg.bletestperipheral&hl=en

closed 2018-01-31 by sandeepmistry