noble/noble #682: Adapter state still unauthorized even after running setcap cap_net_raw+eip

scottsweb, 2017-09-21:
I am running a couple of node tools that use noble. Both are requiring me to run as root despite running: 

`sudo setcap cap_net_raw+eip $(eval readlink -f `which node`)`

Error is:

```
noble warning: adapter state unauthorized, please run as root or with sudo
               or see README for information on running without root/sudo:
               https://github.com/sandeepmistry/noble#running-on-linux
```

The output of `which node` is `/home/scott/.nvm/versions/node/v6.10.2/bin/node`.
The output of `sudo getcap $(eval readlink -f `which node`)` is `/home/scott/.nvm/versions/node/v6.10.2/bin/node = cap_net_raw+eip`

I have also created a symlink for root so the latest version of node is also available at `/usr/local/bin/node`.

I believe my issue may be caused by using [NVM](https://github.com/creationix/nvm) to manage node but cannot be 100% certain. 

Any tips of what I might be able to do to debug this some more?

jrobeson, 2017-09-22:
I'm not yet sure about what your issue really is, but I can say that nvm works fine with noble and that setcap comand.

scottsweb, 2017-09-24:
That is good to know that it is possible to have it working. I will keep investigating.

Erbachm, 2017-10-09:
Just wondering if there has been any progress on this as I'm experiencing the same issue.

scottsweb, 2017-10-09:
No progress from me. I have been unable to find a solution to this yet. For completeness my OS info is:

```
                          ./+o+-       scott@tux
                  yyyyy- -yyyyyy+      OS: Ubuntu 17.04 zesty
               ://+//////-yyyyyyo      Kernel: x86_64 Linux 4.10.0-35-generic
           .++ .:/++++++/-.+sss/`      Uptime: 2m
         .:++o:  /++++++++/:--:/-      Packages: 2461
        o:+o+:++.`..```.-/oo+++++/     Shell: bash 4.4.7
       .:+o:+o/.          `+sssoo+/    Resolution: 1920x1080
  .++/+:+oo+o:`             /sssooo.   DE: Gnome 
 /+++//+:`oo+o               /::--:.   WM: GNOME Shell
 \+/+o+++`o++o               ++////.   WM Theme: Adwaita
  .++.o+++oo+:`             /dddhhh.   GTK Theme: Arc-Darker [GTK2/3]
       .+.o+oo:.          `oddhhhh+    Icon Theme: Paper
        \+.++o+o``-````.:ohdhhhhh+     Font: Cantarell 11
         `:o+++ `ohhhhhhhhyo++os:      CPU: Intel Core i7-7500U CPU @ 3.5GHz
           .o:`.syhhhhhhh/.oo++o`      GPU: Mesa DRI Intel(R) HD Graphics 620 (Kabylake GT2) 
               /osyyyyyyo++ooo+++/     RAM: 6865MiB / 15932MiB
                   ````` +oo+++o\:    
                          `oo++.      
```

jrobeson, 2017-10-09:
have you tried it on a system installed node without nvm? just to get that out of the way?

WayneKeenan, 2017-11-01:
What Linux distro are you on?

This might help you...

I just found out that on Raspbian Stretch, as opposed to previously on Jessie, you now have to add cap_net_admin, i.e.:

```bash
sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f which node)
```

scottsweb, 2017-11-15:
`sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f which node)`

This unfortunately does not seem to help either. I am going to wipe my machine towards the end of December and will see if that does the trick.

WayneKeenan, 2017-11-15:
the node binary is not on an NFS mount (or other FS that doesn't support setcap), is it?

scottsweb, 2017-11-15:
Nope I don't think so. It is in my home folder (which is encrypted).

sandeepmistry, 2018-01-31:
@WayneKeenan could you please support a PR to the docs for this :)

gniezen, 2018-06-18:
I have the same issue and results as above on Ubuntu 18.04. Could be red herrings, but I'm also using nvm and an encrypted home folder on an SSD.

jrobeson, 2018-08-03:
so what does `getcap $(eval readlink -f which node)` say? does it show the capabilities you set?

gniezen, 2018-08-03:
Output of 
```
getcap $(eval readlink -f `which node`)
```
is 
```
/home/gerrit/.nvm/versions/node/v8.2.1/bin/node = cap_net_admin,cap_net_raw+eip
```

I'm executing my program as `node index.js`, which throws above error, and `sudo node index.js` works.

hanhdt, 2018-09-06:
Yes, I also have the same issue and results as above on Ubuntu 18.04.

For me `getcap $(eval readlink -f `which node`)`
output is `/usr/local/bin/node = cap_net_admin,cap_net_raw+eip`

I'm working on an Electron app.

jrobeson, 2018-09-06:
@hanhdt that won't work on an electron app because the executable isn't node.  you have to change it to point to your electron binary.

ctangc5, 2018-12-03:
Dears,

I also have the same issue. > sudo setcap cap_net_raw+eip /usr/lib/node_modules/node-red-contrib-xiaomi-ble
Output is: Failed to set capabilities on file ‘usr/lib/node_modules/node-red-contrib-xiaomi-ble’ (Invalid argument)
The value of the capability argument is not permitted for a file. Or the file is not a regular (non-symlink) file

cropab, 2018-12-21:
@ctangc5

after running this command: 

sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f `which node`)

the noble permission warning disappeared and after a restart node-red-contrib-xiaomi-ble started reading my mijia sensor.

zhaojjxvi, 2018-12-22:
> @ctangc5
> 
> after running this command:
> 
> `sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f which node)`
> 
> the noble permission warning disappeared and after a restart node-red-contrib-xiaomi-ble started reading my mijia sensor.
> 
> ps: this will only work in node red with node 8.

I was encoutering the same issue as yours, it's working well now after i execute the command

`sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f `which node`)`

ctangc5, 2018-12-24:
@cropab, zhaojjxvi,
I still have the issue follow your step. Below is my steps for reference.

pi@raspberrypi:~ $ node -v
v8.11.1
pi@raspberrypi:~ $ cd /usr/lib/node_modules/
pi@raspberrypi:/usr/lib/node_modules $ sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f node-red-contrib-xiaomi-ble)
Failed to set capabilities on file `/usr/lib/node_modules/node-red-contrib-xiaomi-ble' (Invalid argument)
The value of the capability argument is not permitted for a file. Or the file is not a regular (non-symlink) file
pi@raspberrypi:/usr/lib/node_modules $ sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f "node-red-contrib-xiaomi-ble")
Failed to set capabilities on file `/usr/lib/node_modules/node-red-contrib-xiaomi-ble' (Invalid argument)
The value of the capability argument is not permitted for a file. Or the file is not a regular (non-symlink) file
pi@raspberrypi:/usr/lib/node_modules $ dir
node-red                         node-red-contrib-xiaomi-ble
node-red-contrib-bacnet          node-red-contrib-yeelight
node-red-contrib-ibm-watson-iot  node-red-dashboard
node-red-contrib-knx             node-red-node-ledborg
node-red-contrib-modbus          node-red-node-ping
node-red-contrib-mqtt-broker     node-red-node-pi-sense-hat
node-red-contrib-play-audio      node-red-node-random
node-red-contrib-tensorflow      node-red-node-serialport
node-red-contrib-viseo-wechat    node-red-node-smooth
pi@raspberrypi:/usr/lib/node_modules $ sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f 'node-red-contrib-xiaomi-ble')
Failed to set capabilities on file `/usr/lib/node_modules/node-red-contrib-xiaomi-ble' (Invalid argument)
The value of the capability argument is not permitted for a file. Or the file is not a regular (non-symlink) file
pi@raspberrypi:/usr/lib/node_modules $

ctangc5, 2018-12-24:
@zhaojjxvi,
 In my case, 'which node' is "/usr/lib/node_modules/node-red-contrib-xiaomi-ble". I use the following command and still have issue.
> sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f /usr/lib/node_modules/node-red-contrib-xiaomi-ble)
Failed to set capabilities on file `/usr/lib/node_modules/node-red-contrib-xiaomi-ble' (Invalid argument)
The value of the capability argument is not permitted for a file. Or the file is not a regular (non-symlink) file

cropab, 2018-12-24:
first i tried `sudo setcap cap_net_raw+eip /usr/lib/node_modules/node-red-contrib-xiaomi-ble` didnt work, same error messages. 

after that i issued `sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f which node)` , just like it is, and there was no console output to a path or anything. ran the command, restarted node red and the mijia was sending data the first time.

OS: 2018-11-13-raspbian-stretch-full.img with preinstalled Node Red, Node 8.

ctangc5, 2018-12-26:
@WayneKeenan, when I run "sudo getcap /usr/lib/node_modules/node-red-contrib-xiaomi-ble" without any feedback.

ctangc5, 2018-12-26:
@cropab, Thanks your guide. Below is my steps but still have issue.
Setp 1. >sudo setcap cap_net_raw+eip /usr/lib/node_modules/node-red-contrib-xiaomi-ble
Failed to set capabilities on file `/usr/lib/node_modules/node-red-contrib-xiaomi-ble' (Invalid argument)
The value of the capability argument is not permitted for a file. Or the file is not a regular (non-symlink) file

Step 2. >sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f which node)
Failed to set capabilities on file `/home/pi/which' (No such file or directory)
The value of the capability argument is not permitted for a file. Or the file is not a regular (non-symlink) file

P.S. My node.js is 8.11.1 also use 2018-11-13-raspbian-stretch-full.img

WayneKeenan, 2018-12-26:
@ctangc5 after you run that what does running : echo $?

Show?

ctangc5, 2018-12-28:
@WayneKeenan,  below is my command running
>sudo echo $
feedback display "$"
>sudo echo $?
feedback display "0"

WayneKeenan, 2018-12-28:
@ctangc5 please run the following exactly as is:

```shell
sudo getcap /usr/lib/node_modules/node-red-contrib-xiaomi-ble
echo $?
```

The `echo $?` shows the return code of the previous shell command, where 0 means success.

ctangc5, 2018-12-29:
@WayneKeenan, follow your instruction the feedback is "0" However I also command "sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f /usr/lib/node_modules/node-red-contrib-xiaomi-ble)" the echo $? shows the "1"

pdfrod, 2019-01-31:
Is the binary inside an encrypted folder? As explained [here](https://stackoverflow.com/a/30826137/409157) Linux capabilities have no effect if the file is on an encrypted folder.

scottsweb, 2019-01-31:
That could have well been my issue then, I mentioned the encrypted drive way back here @pdfrod: https://github.com/noble/noble/issues/682#issuecomment-344589269 - I am no longer working on the project that uses this library and my computer setup has changed since. 

I am happy to close this now if there are no objections?

jrobeson, 2019-01-31:
not all ways of encrypting folders are the same.  Entire partition encryption (which could also be called encrypted /home) is fine, but probably not when using encryptfs.

scottsweb, 2019-01-31:
I only had my home folder encrypted I believe and `encryptfs` sounds super familiar 👍

IamfromSpace, 2019-08-02:
Hey all, I ran into this same issue and it also turned out to be caused by the encryptfs issue https://bugs.launchpad.net/ecryptfs/+bug/1147348

My home directory is encrypted, and I use nvm, which by default stores the node binary in the home directory.  I can see the capabilities are set, but they only work if I move the binary file outside of the encrypted dir.

rzr, 2020-09-06:
Relate-to: https://github.com/noble/noble/issues/923#issuecomment-687851954

rzr, 2021-01-26:
It looks the maintenance of this project is in pause, but feel free to contribute to @abandonware fork until upstream is active again

Feel free to comment:
https://github.com/abandonware/noble/issues/2

JachiOnuoha, 2022-04-24:
> 

I had a similar issue when trying to use gattacker on Kali Linux for ARM. This resolved it for me