noble/noble #683: Battery level

tsadak, 2017-09-21:
How can I read the battery level of a beacon (classified as ibeacon)?

cescocesco, 2017-09-28:
Hi tsadak

I use this code and works fine

`noble.on('discover', function(peripheral) {
  peripheral.connect(function(error) {
    console.log('connected to peripheral: ' + peripheral.uuid);
    peripheral.discoverServices(['180f'], function(error, services) {
      var batteryService = services[0];
      console.log('discoveredBatter service');

      batteryService.discoverCharacteristics(['2a19'], function(error, characteristics) {
        var batteryLevelCharacteristic = characteristics[0];
        console.log('discovered Battery Level characteristic');

        batteryLevelCharacteristic.on('data', function(data, isNotification) {
          console.log('battery level is now: ', data.readUInt8(0) + '%');
        });

        // to enable notify
        batteryLevelCharacteristic.subscribe(function(error) {
          console.log('battery level notification on');
        });
      });
    });
  });
});`

You can find more information on wiki here: [https://github.com/sandeepmistry/noble/wiki/Getting-started](url)

tsadak, 2017-09-28:
Thank you for answering.

 

Which beacon did you use? I am trying this code with a mobile phone beacon simulator, but no answer from any type of beacon.

cescocesco, 2017-09-28:
@tsadak i use a bluetooth tag by Amicom, similar to this [https://www.amazon.it/XinXu-Localizzatore-Bluetooth-Bidirezionale-Allarmante/dp/B01KLNM5OE/ref=sr_1_16?ie=UTF8&qid=1506602158&sr=8-16&keywords=tag+bluetooth](url) . "180f" is standard for Battery service while characteristic "2a19" is battery level the output is hex (64 = 100%). [https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.battery_level.xml&u=org.bluetooth.characteristic.battery_level.xml](url)

I suggest you to use first interactive tool like gatttool and after create a js code. Here is a clear tutorial [http://blog.firszt.eu/index.php?post/2015/09/13/bt](url)

closed 2018-01-31 by sandeepmistry