noble/noble #685: Can't subscribe to characteristics

aplyumpv, 2017-09-24:
Hello,

It seems that subscription to characteristics does not work as "Characteristic found..." is never displayed.
(same issue as #663 only with correct source version)
Could anyone please check this?

```javascript
var noble = require('noble');

noble.on('stateChange', function (state) {
  if (state === 'poweredOn') {
    noble.startScanning();
  } else {
    noble.stopScanning();
  }
});

noble.on('discover', (peripheral) => {
  if (peripheral.id.toUpperCase() == 'B2F669A94E45') {
    console.log('Connected...');
    peripheral.connect((err) => {
      peripheral.discoverServices(['49535343fe7d4ae58fa99fafd205e455'], (err, services) => {
        console.log("Service discovered...");
        services.forEach((service) => {
          service.discoverCharacteristics(['495353431e4d4bd9ba6123c647249616'], (err, characteristics) => {
            console.log('Characteristic found...');
            var characteristic = characteristics[0];
            characteristic.on('data', (data) => {
              characteristic.read((err, readData) => {
                console.log(readData.toString('utf-8'));
              });
            });
            characteristic.subscribe(()=> {
				console.log('Subscribed to characteristic ...');
			});
          });
        });
      });
    });
  }
});
```

cescocesco, 2017-09-27:
I have same problem. I try characteristic.notify(true,function(err) { ... }); but doesn't work

zhaiyuzhai, 2018-05-10:
I have the same problem. Have you solved the problem?And how

VojtechStep, 2019-02-08:
Any updates on this? Can't get this library working with the RN4871 module. Sending works fine, but no luck on receiving.

ns16, 2019-08-21:
I have the same problem.