noble/noble #686: How to use effectively?

ProbablePrime, 2017-09-24:
Great library, I'm just a little confused on how to use it the most effectively?

My scenario involves writing some values to a characteristic in response to Node.JS sourced events.

My initial attempt was to setup a constant connection to the BLE Device:
```
connect 
discover services and characteristics
on disconnect repeat
```

```
on nodejs event
use active connection from first block
write characteristic
```

This seems to be against the whole point of BLE so I'm now looking for the correct way to handle this scenario which appears to be:
```
receive nodejs event
connect
write characteristic
disconnect
```

The problem here is in discovery which takes a long time each cycle. From what I can read there seems to be no reason to discover all of the time but the module isn't set up in a way that remembers the discovered structures. For example `peripheral.services` is null and never written to.

Questions:
1. is my new scenario the correct way to do this?
1. Is discovery meant to be performed once per peripheral discovery? 
1. Are discovered services meant to be cached? (I did notice https://github.com/sandeepmistry/noble/pull/559 which seems to be what i'm looking for but am having no luck.)

jrobeson, 2017-09-28:
having no luck after applying  #559 ?

ProbablePrime, 2017-09-28:
Nope no luck after that. I'll update the issue with os and stuff thanks

sandeepmistry, 2018-01-31:
Discover should be called each time after connecting.

closed 2018-01-31 by sandeepmistry