noble/noble #687: Multiple instances in Windows problem

jsantos98, 2017-09-26:
Hi,

While trying to have multiple instances of a noble connection, I'm always getting an exception.

To reproduce:
1. Open a command window
2. browse to the examples folder
3. run "node advertisement-discovery.js"
4. Leave that window running, open a new one
5. Browse again to the examples folder
6. Run the same "node advertisement-discovery.js"

I'm getting the following exception:

> ...\node_modules\usb\usb.js:33
>         this.__open()
>              ^
> 
> Error: LIBUSB_ERROR_ACCESS
>     at Error (native)
>     at Device.usb.Device.open (N:\IoMT\src\driver-ble\node_modules\usb\usb.js:33:7)
>     at BluetoothHciSocket.bindUser (N:\IoMT\src\driver-ble\node_modules\bluetooth-hci-socket\lib\usb.js:73:19)
>     at Hci.init (N:\IoMT\src\driver-ble\node_modules\noble\lib\hci-socket\hci.js:96:35)
>     at NobleBindings.init (N:\IoMT\src\driver-ble\node_modules\noble\lib\hci-socket\bindings.js:82:13)
>     at new Noble (N:\IoMT\src\driver-ble\node_modules\noble\lib\noble.js:50:18)
>     at Object.<anonymous> (N:\IoMT\src\driver-ble\node_modules\noble\index.js:4:18)
>     at Module._compile (module.js:570:32)
>     at Object.Module._extensions..js (module.js:579:10)
>     at Module.load (module.js:487:32)
> 

Any idea how to bypass this?

Thanks

lemmerelassal, 2017-09-27:
With LibUSB you can only claim the interface once.

jsantos98, 2017-09-27:
I was already suspecting for a restriction like that :(

Is there any workaround to support multiple instances?

Regards
Joao

lemmerelassal, 2017-09-27:
Oh, you want to connect to multiple devices via one dongle, eh?
This is doable using one script alone!

sandeepmistry, 2018-01-31:
@jsantos98 use multiple USB adapters.

@lemmerelassal thanks for helping out!

closed 2018-01-31 by sandeepmistry