noble/noble #688: notify

cescocesco, 2017-09-26:
I try to read button pressed on bluetooth tag. Battery level works great either with gatttool and noble, but i don't know how to reproduce this command 

gatttool -b xx:xx:xx:xx:xx:xx —char-write-req —handle=0x0053 —value=0100 —listen

The equivalent uuid of handle 0x0053 is FFF1 : 
 characteristic:{"uuid":"fff1","name":null,"type":null,"properties":["notify"]}

I'll try

bt1.on('data',function (error,data) {
                                console.log(data);
                        });
                        bt1.subscribe( function(error) {
                                console.log('button notification on');
                        });

Also i try to send before data, or put data and subscribe inside this function:

bt1.write(setVal,true, function (err) { });

Any solution? Thanks

cescocesco, 2017-09-28:
> gatttool -b xx:xx:xx:xx:xx:xx —char-write-req —handle=0x0053 —value=0100 —listen
> The equivalent uuid of handle 0x0053 is FFF1 :
>characteristic:{"uuid":"fff1","name":null,"type":null,"properties":["notify"]}

I make mistake to indicate the equivalent uuid of handle 0x0053. Actually i must write value 0100 to handle 0x0053 but the response handle is 0x0052 that is FFF1 uuid

> bt1.subscribe( function(error) {
bt1 is the characteristic uuid FFF1 of FFF0 service. I try to subscribe or write Buffer.from([0x100]) but nothing happened either with .on('data' or .on('notify' .

When i try with ble utility app on my phone the characteristic FFF1 listen correctly and button pressed notify correctly.

Help me :) ... Thanks for any answer!

cescocesco, 2017-09-29:
I'm able to fix in this way:

`buttonService.discoverCharacteristics(['fff1'], function(error, c) {
    	      var bt1 = c[0];
            bt1.on('data',function (data,idNotification) {
              console.log('button read',data.readUInt8(0));
            });
    	      peripheral.writeHandle(0x0053,Buffer.from('0100','hex'),false,function (error) { 
console.log('button active'); });
          });`

Without use "subscribe" method. # #

aplyumpv, 2017-09-30:
This seems to work for me too despite standard subscribe didn't (#685) . Thanks.

cescocesco, 2017-09-30:
Nice @aplyumpv !

sandeepmistry, 2018-01-31:
Closing as duplicate of https://github.com/sandeepmistry/noble/issues/685.

closed 2018-01-31 by sandeepmistry