noble/noble #692: Reconnection

cescocesco, 2017-09-30:
I have:

noble.on('discover', function(peripheral) {
peripheral.connect(function(error) {
 peripheral.once('disconnect',function () { console.log('disconnect'); noble.startScanning([], true); });
 ... <some discover service>
 ... .on('data', function () { console.log('data'); });
});
});

When the peripheral disconnect for the second time, the event on('data') fired twice and the script don't work correctly.

Is there any fix or any solution?

Thanks

cescocesco, 2017-10-10:
I solved by put noble._peripherals = [] once disconnect. Seems disconnect don't empty data of previous discover.

ryter--, 2018-01-05:
Wow, thanks a lot! I had the same issue and your solution worked wonderfully. How did you discover your solution?

cescocesco, 2018-01-05:
@ryter-- glad to help you. I see that when reconnection happen, some function was called twice or multiple time. After i debug some var and discover this solution.

closed 2018-01-31 by sandeepmistry