noble/noble #693: Packets can be received by Linux but not Mac

zhiheng-zhang, 2017-10-03:
I have a packet which format like {0x02, 0x01, 0x1A, 0x1A, 0xFF, 0x4C, 0x00, 0x02, 0x15, 0x40, 0x50, 0x40, 0x40, 0x40, 0x40, 0x40} by a embedded LOT device, and I'm trying to get it received by both Mac and Linux machine by noble. When I run the code advertisement_discovery.js, my linux machine can receive my packets but my Mac machine couldn't discover it. What is the potential reason that it could possibly be? 

I know Linux and Mac are using different stacks of BLE, but I'm not sure what could cause this problem? Thanks for any hints?

Axadiw, 2017-10-09:
Are you using MacOS HighSierra? Support for this system was added in this PR: https://github.com/sandeepmistry/noble/pull/689

zhiheng-zhang, 2017-10-13:
@Axadiw  Cheers man! The support file really works! But I'm coming to my next trouble which is: how can I get the row packets (in bytes) from highsierra.js? I'm looking for the row packets which are sending from my LoTs rather than some structured packets leading by kCBMsg, but the xpcEvent is the most I can backtrace.

closed 2018-03-12 by sandeepmistry