noble/noble #698: Using Noble on Raspberry Pi 3 doesnt work vs. on Mac

teuteuguy, 2017-10-11:
Hi, 

I'm trying to run Noble on raspberry pi to develop a communication system between a bleno device (edison) and Rapsberry Pi (Noble).

Problem is that my code, on Raspberry Pi doesnt work, where it does work on my Mac.

Not sure what is going on.
The service/characteristics of the peripheral are not getting picked up by the raspberry pi.

Here's my code: 
var noble = require('noble');

var uuid_device = 'e2c56db5dffb48d2b060d0f5a71096e1';
var uuid_charac = 'e2c56db5dffb48d2b060d0f5a71096e2';

noble.on('stateChange', function (state) {
    if (state === 'poweredOn') {
        console.log('scanning...');
        noble.startScanning([uuid_device], false);
    }
    else {
        noble.stopScanning();
    }
});

noble.on('discover', function (peripheral) {
    // found a peripheral, stop scanning
    noble.stopScanning();

    console.log('found peripheral:', peripheral.advertisement);

    peripheral.connect(function (err) {
        console.log('connect', err);

        peripheral.discoverAllServicesAndCharacteristics(function(error, services, characteristics) {
            console.log('discoverAllServicesAndCharacteristics', error);
            console.log(services);
            console.log(characteristics);
        });

Result on the Pi:
node index.js
scanning...
found peripheral: { localName: 'EdisonAccel',
  txPowerLevel: undefined,
  manufacturerData: undefined,
  serviceData: [],
  serviceUuids: [ 'e2c56db5dffb48d2b060d0f5a71096e1' ],
  solicitationServiceUuids: [],
  serviceSolicitationUuids: [] }
connect null
... and then nothing.

Result on the Mac with exact same code:
node index.js
scanning...
found peripheral: { localName: 'timsedison',
  txPowerLevel: undefined,
  manufacturerData: undefined,
  serviceData: [],
  serviceUuids: [ 'e2c56db5dffb48d2b060d0f5a71096e1' ] }
connect undefined
discoverAllServicesAndCharacteristics null
[ Service {
    _noble:
     Noble {
       state: 'poweredOn',
       address: 'a4:5e:60:ea:be:54',
       _bindings: [Object],
       _peripherals: [Object],
       _services: [Object],
       _characteristics: [Object],
       _descriptors: [Object],
       _discoveredPeripheralUUids: [Object],
       _events: [Object],
       _eventsCount: 3,
       _allowDuplicates: false },
    _peripheralId: 'f53f7e7954a24c859098cd0b75c95f58',
    uuid: 'e2c56db5dffb48d2b060d0f5a71096e1',
    name: null,
    type: null,
    includedServiceUuids: null,
    characteristics: [],
    _events: {},
    _eventsCount: 0 } ]
[]

homedad, 2017-10-12:
Have you run your node script as sudo on the pi? Else you won't get access to BLE. You can also see this with hcitool. `hcitool scan` works, but sudo is required, for lescan: `sudo hcitool lescan`. The readme for this project tells how to use setcap to enable non-sudo access to ble.

sandeepmistry, 2018-01-31:
Closed via https://github.com/sandeepmistry/noble/pull/727.

closed 2018-01-31 by sandeepmistry