noble/noble #699: notify never callback on windows

advanpro, 2017-10-11:
CSR8510 A10

peripheral.discoverSomeServicesAndCharacteristics(serviceFilters, characteristicFilters, (error, services, characteristics) => {
         
          characteristics.forEach((element) => {
         
            if (element.properties.find((item) => item === 'notify')) {
              console.log('subscribe',  element.uuid)
              element.subscribe(() => console.log('callback', element.uuid))
            }
          })
        })

noble warning: unknown handle 72 disconnected!
poweredOn
subscribe 808f
subscribe 2a19
subscribe 00001c0fd10211e19b23000efb00dbdb
subscribe 00001c0ed10211e19b23000efb00dbdb
<Buffer 09 02 24 00>
<Buffer 09 04 3a 00 00 00>
<Buffer 01 08 2f 00 02>
<Buffer 01 08 34 00 02>
callback 808f
callback 2a19

the uuid of 128 bit always not callback.

ronangaillard, 2017-11-15:
I confirm that using Mac OSX Sierra and 128 uuid, subscribing is not working (notification never trigger callback)

zhaiyuzhai, 2018-05-10:
on('data');notify();subscribe(),every method i have tried .But,not working!anyone solved?

LorenzGardner, 2018-05-14:
I had/have the same issue on windows. The solution for me was to turn on the debug prints, and then everything just started working.

set DEBUG=*,-not_this

As you can find:
https://www.npmjs.com/package/debug

I'm guessing this changes the timing just enough to allow the driver to behave more consistent. After doing this everything worked better. Before hand I'd have to stop and restart several times to get the powered on state in noble and the advertisement call to work. But then always hung on discovery during connection. After setting the debug prints, it all worked without issue. 

Anyway, don't know if that is helpful or a clue to what is going on.