noble/noble #701: Passive scanning with noble

someone-noone, 2017-10-15:
In my project, I need to perform passive scanning (without scan req\scan resp). I forked the repository and created a workaround for this for HCI socket. However, it would be nice to have such API in noble. 

What's an appropriate way to provide such configuration in noble? Environment variable? Is it possible to do the passive scanning on OS X? I could submit a pull request.

FROeHlyEisvogel, 2018-01-17:
Hey there,
sorry that I'm pushing this post but I have the same problem as described above.
I have an beacon which sould be tracked but the active scanning mode is too slow.
I have already build a whole plugin for noble after noticing this.
It would be a shame if I have to rebuild it with using hcitool/hcidump.

Could you please implement the option for passive scanning or call me how to perform a passive scan using noble.

Thank you for this very nice software.

someone-noone, 2018-01-17:
It's very simple, check this out:

https://github.com/someone-noone/noble/commit/7e0d521d0a7e029406ee709fe418210f6c464e78

If an interface for this could be suggested, I could submit a pull-request.

sandeepmistry, 2018-01-31:
@someone-noone cool, a pull request for env var seems good.

This is not possible on macOS.

lcrocker, 2018-07-16:
As suggested by someone-noone, you can change the scan parameters to passive inside hci.js, but then I stop getting the 'discover' events entirely.