noble/noble #702: Linux: Using Bleno and Noble at the same time by switch state

zhiheng-zhang, 2017-10-18:
Hi, I'm currently working on a application which needs to receive and transmit BLE packet using one BLE adapter. I've realised noble and bleno cannot be turning on at the same time, but my application only requires most of the time it does a noble so it keep receiving packet. Only when it get some data from socket, it needs to advertising packets. So the thing I want to do is: when it receive some data from socket, disable noble and enable bleno, advertising packet for 2 seconds, and then disable bleno and switch noble back. Is this any solution for how to achieve it ?

zhiheng-zhang, 2017-10-24:
Achieved this functionality by connect/disconnect hcisocket by bleno and noble in turn.

closed 2017-10-24 by zhiheng-zhang

RasPelikan, 2020-01-23:
> Achieved this functionality by connect/disconnect hcisocket by bleno and noble in turn.

Could you please past a code sample or a link if it is in github? I need achieve the same and do not understand how you implemented this. You would save me a lot of time. Thank you in advance!