noble/noble #703: WriteWithoutNotifications trigger buffer overflows in the controller

IvanSanchez, 2017-10-18:
TL;DR: Sending several (~8) commands to writeWithoutNotification to a characteristic in quick succession triggers a buffer overflow in the controller. The buffer size and usage can be handled but is currently not. The end result is the HCI adapter crashing.

---

My environment is a Linux host (debian testing, kernel 4.13, bluez 5.47) with an Intel 8275 dual band PCIe card. The problem can be reproduced in very similar circumstances with an CSR8510A10 USB dongle; my analysis suggests that the problem reproduces in other platforms as well under the right conditions.

The scenario is performing a DFU (Device Firmware Upgrade) on a [nRF 52832 development kit board](https://www.nordicsemi.com/eng/Products/Bluetooth-low-energy/nRF52-DK) using Nordic's [BLE DFU protocol](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v14.0.0%2Flib_dfu_transport_ble.html&cp=4_0_0_3_5_2_2) (using the firmware images available in the nordicsemi SDK 14.0.0). This protocol involves sending a big (~100KiB) binary blob by means of repeated write calls to a characteristic with the writeWithoutNotification flag set.

I've done my best to provide a [minimal example triggering this bug](https://gist.github.com/IvanSanchez/7e4d796616f4dfd85598bf0b1d627a25). Note that I'm sending write commands as soon as I'm notified of the previous command success thanks to a bit of magic with `Promise`s (code extracted from my gist, slightly simplified):

```js
// Returns a Promise
function sendBytes(endpoint, bytes) {
    return new Promise(function(res, rej){
            endpoint.write(bytes, true, function(err) { // Note writeWithoutNotification is true
                if (err) rej(err); else res();
            });
    });
}

// Sends a bunch of data blocks and then waits for a notification.
function sendDataBlock(dataEndpoint, prnLeft) {
    return sendBytes(dataEndpoint, new Buffer([1,2,3,4,5,6,7,8]))
    .then(function(){
        return sendDataBlock(dataEndpoint, prnLeft - 1);
    })
}
```

---

The issue is that the bluetooth controller has a *quite small* buffer for packets pending to be sent. By attaching Wireshark, I can see 0x05 ("Read Buffer Size") and 0x2002 ("LE Read Buffer Size") being sent, as well as their responses. The responses indicate that the buffer size is about 4 for the intel adapter, and about 8 for the CRS8510. This buffer size impacts how this bug is triggered.

Whenever a packet is written to the Controller, it's not sent into the air - instead it goes into that small transmit buffer. Whenever the packet actually goes into the air, the Host receives a HCI event `0x13` ("Number of completed packets") indicating that the buffer is being flushed. It can also receive a `0x1A` ("Data Buffer Overflow") event. 

As far as I can tell, writing to a characteristic with `writeWithoutNotification` set to `false` will wait until the response is received ("wait" meaning "calling the callback"). But writing to a characteristic with `writeWithoutNotification` set to `true` does **not** wait until the packet goes into the air. See this wireshark screenshot:

[image]

The four topmost packets are the usual data flow: host asks controller to send some data, controller notifies that the packet is in the air (after ~0.11sec), response is received (after another ~0.11sec). Note that there is also a delay of ~0.10sec between the first `writeWithoutResponse` command and the first corresponding  `0x13` ("Number of completed packets") notification, but `noble` sends all write operations with practically no delay.

In this bug, `noble` is not aware of the state of the controller's TX queue/buffer, and is not listening to the overflow event, so it keeps sending data to the controller, the controller crashes, `noble` keeps running without any error notification, no data can be read anymore, the rest of the program goes awry. Interestingly enough, the Intel controller crashes by entering an infinite loop, and the CRS crashes by resetting and powering off.

My understanding of the [`/lib/hci-socket/hci.js` file](https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/hci.js#L16) is that `noble` is not even subscribing to the `0x13` ("Number of completed packets") event, and not caring about the number of elements in the controller's data buffer at all.

I would suggest starting the bugfixing by subscribing to the events related to this (buffer size, buffer overflow, number of completed packets),  just to print debug information when they are handled. Then, a counter of how many items are there in the controller's buffer could be implemented. I would further guess that the optimal solution goes through using a [`tinyqueue`](https://github.com/mourner/tinyqueue) with the packets pending to be sent to the controller.

gotleg, 2018-01-22:
@IvanSanchez Very interesting. I'm dealing with a similar behavior.
I'm writing characteristics every 100ms to a peripheral and if I disconnect my peripheral (unplug battery) my CSR dongle crash and goes poweredOff (probably because of full buffer).
The disconnect event comes to late so I can't stop my writing. At 500ms it is ok and dongle doesn't crash (probably beacause buffer is not full yet.).

Did you go further finding a solution ?

gotleg, 2018-01-25:
@IvanSanchez I coded a workaround in my fork https://github.com/gotleg/noble
No more crash with my CSR dongle.
Max packet is hardcoded to 8 for the moment because I didn't have success with "LE Read Buffer Size" but I will try to go further soon.

sandeepmistry, 2018-01-31:
We need a port of https://github.com/sandeepmistry/bleno/pull/268 to noble.

Pull requests welcome for this :)