noble/noble #704: Nothing happens once examples are run

simonem, 2017-10-20:
Just installed noble on OSX High Sierra.
Tried to run the example but nothing shows up launching advertisement-discovery.js or enter-exit.js
I get a "Cannot read property 'toLowerCase' of undefined" launching peripheral-explorer.js
I have a few BTLE devices (RFDUINO and nRF52) discoverable nearby.

Thanks
Simone

fivdi, 2017-10-22:
@simonem 

> I get a "Cannot read property 'toLowerCase' of undefined" launching peripheral-explorer.js
I have a few BTLE devices (RFDUINO and nRF52) discoverable nearby.

A `Cannot read property 'toLowerCase' of undefined` error is thrown when no peripheral ID or address is passed on the command-line when running peripheral-explorer.js.

Passing a peripheral address on the command-line looks something like this:
```
sudo node peripheral-explorer.js 12:34:56:78:9A:BC
```
Passing a peripheral ID on the command-line looks something like this:
```
sudo node peripheral-explorer.js 123456789ABC
```

simonem, 2017-10-24:
@fivdi Thank you!

closed 2017-10-24 by simonem