noble/noble #707: cannot install noble on RPI3

lachlan-peter, 2017-11-01:
Have RPI3 running latest OS (Linux 4.9) [Stretch-lite) and latest nodejs (8.9 LTS) and loaded:
$ sudo apt-get install bluetooth bluez libbluetooth-dev libudev-dev bluez build-essential

and trying to install noble using:
$ sudo npm install -g noble

Fails with:
WARN checkPermissions Missing write acccess to /home/pi/noble/node_modules/xpc-connection
WARN checkPermissions Missing write access to /home/pi/noble/node_modules
ERR path /home/pi/noble/node_modules/xpc-connection

Dont know why it is installing in local directory and not in /usr/lib/node_modules
Also dont know how to overcome the permissions problem.

Tried with the --build-from-source option and getting lots of permission denied errors. This at least put the module in /usr/lib/node_modules.

Also tried downloading from git
$ git clone https://github.com/sandeepmistry/noble.git
$ cd noble
create bindings file with source -> index.js
$ node-gyp configure
$ node-gyp build
But also get errors

Others seem to be running noble on rpi ok. Maybe its because I am running a minimal version of 'stretch' and something is missing. Maybe noble is not compatible with node 8.9 LTS. Other node modules load ok such as socket.io, node-red, and others. Serialport i have to compile for armhf machine but works ok.

Any suggestions on where to look?
Lachlan

seppestas, 2017-11-02:
Why are you installing noble globally?

lachlan-peter, 2017-11-03:
I was proposing to use noble for node-red applications and also my own node applications.

-------------------------------------------------------------
However, I have tried to install noble locally in the ~/.node-red directory but get the error:
~/.node-red $ npm install noble
npm WARN checkPermissions Missing write access to /home/pi/.node-red/node_modules/noble
npm WARN node-red-project@0.0.1 No repository field.
npm WARN node-red-project@0.0.1 No license field.

npm ERR! path /home/pi/.node-red/node_modules/noble
npm ERR! code EACCES
npm ERR! errno -13
npm ERR! syscall access
npm ERR! Error: EACCES: permission denied, access '/home/pi/.node-red/node_modules/noble'

-------------------------------------------------------------------
When I try to install just in the home directory I get the error:
~ $ npm install noble
> usb@1.2.0 install /home/pi/node_modules/usb
> node-pre-gyp install --fallback-to-build
node-pre-gyp info it worked if it ends with ok
node-pre-gyp verb cli [ '/usr/bin/node',
node-pre-gyp verb cli   '/home/pi/node_modules/usb/node_modules/.bin/node-pre-gyp',
node-pre-gyp verb cli   'install',
node-pre-gyp verb cli   '--fallback-to-build' ]
node-pre-gyp info using node-pre-gyp@0.6.30
node-pre-gyp info using node@8.9.0 | linux | arm
node-pre-gyp verb command install []
node-pre-gyp info check checked for "/home/pi/node_modules/usb/src/binding/usb_bindings.node" (not found)
node-pre-gyp http GET https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v57-linux-arm.tar.gz
node-pre-gyp http 404 https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v57-linux-arm.tar.gz
node-pre-gyp ERR! Tried to download: https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v57-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for usb@1.2.0 and node@8.9.0 (node-v57 ABI) (falling back to source compile with node-gyp)
node-pre-gyp http Pre-built binary not available for your system, looked for https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v57-linux-arm.tar.gz

Followed by lots of errors:

An ideas of the problem?
Lachlan

seppestas, 2017-11-03:
I think you might have SD card corruption problems, sometimes this causes the SD card to become unwritable which might be reported as a permission error.

Could you try just creating a file in your home directory?

lachlan-peter, 2017-11-04:
I have no problems installing other npm modules locally and globally, so there appears to be no problem with the SD  card. Also, because I have spent a lot of time trying to compile the code from source and installing many different ways, I re-imaged my SD card with a new image and I am still having problems with only this module whereas all other functions such as apt-get update and apt-get install work fine.

forrestal, 2018-01-03:
@lachlan-peter have you found solution tho this problem? Experiencing exactly the same behavior on my rpi3

grover, 2018-01-03:
Please try installing using ```npm install -g noble --unsafe-perm``` - that solved the gyp issues in my case.

forrestal, 2018-01-04:
Ok, got some better results

```pi@raspberrypi:~ $ sudo npm install -g noble --unsafe-perm

> usb@1.3.1 install /usr/lib/node_modules/noble/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/tessel/node-usb/releases/download/1.3.1/usb_bindings-v1.3.1-node-v51-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for usb@1.3.1 and node@7.10.1 (node-v51 ABI, glibc) (falling back to source compile with node-gyp)
make: Entering directory '/usr/lib/node_modules/noble/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
  AR(target) Release/obj.target/usb.a
  COPY Release/usb.a
  CXX(target) Release/obj.target/usb_bindings/src/node_usb.o
  CXX(target) Release/obj.target/usb_bindings/src/device.o
  CXX(target) Release/obj.target/usb_bindings/src/transfer.o
  SOLINK_MODULE(target) Release/obj.target/usb_bindings.node
  COPY Release/usb_bindings.node
  COPY /usr/lib/node_modules/noble/node_modules/usb/src/binding/usb_bindings.node
  TOUCH Release/obj.target/action_after_build.stamp
make: Leaving directory '/usr/lib/node_modules/noble/node_modules/usb/build'
/usr/lib
└─┬ noble@1.8.1
  └─┬ bluetooth-hci-socket@0.5.1
    └─┬ usb@1.3.1
      └─┬ node-pre-gyp@0.6.39
        ├── detect-libc@1.0.3
        ├─┬ hawk@3.1.3
        │ ├── boom@2.10.1
        │ ├── cryptiles@2.0.5
        │ ├── hoek@2.16.3
        │ └── sntp@1.0.9
        ├─┬ mkdirp@0.5.1
        │ └── minimist@0.0.8
        ├─┬ nopt@4.0.1
        │ ├── abbrev@1.1.1
        │ └─┬ osenv@0.1.4
        │   ├── os-homedir@1.0.2
        │   └── os-tmpdir@1.0.2
        ├─┬ npmlog@4.1.2
        │ ├─┬ are-we-there-yet@1.1.4
        │ │ └── delegates@1.0.0
        │ ├── console-control-strings@1.1.0
        │ ├─┬ gauge@2.7.4
        │ │ ├── aproba@1.2.0
        │ │ ├── has-unicode@2.0.1
        │ │ ├── object-assign@4.1.1
        │ │ ├── signal-exit@3.0.2
        │ │ ├─┬ string-width@1.0.2
        │ │ │ ├── code-point-at@1.1.0
        │ │ │ └─┬ is-fullwidth-code-point@1.0.0
        │ │ │   └── number-is-nan@1.0.1
        │ │ ├─┬ strip-ansi@3.0.1
        │ │ │ └── ansi-regex@2.1.1
        │ │ └── wide-align@1.1.2
        │ └── set-blocking@2.0.0
        ├─┬ rc@1.2.2
        │ ├── deep-extend@0.4.2
        │ ├── ini@1.3.5
        │ ├── minimist@1.2.0
        │ └── strip-json-comments@2.0.1
        ├─┬ request@2.81.0
        │ ├── aws-sign2@0.6.0
        │ ├── aws4@1.6.0
        │ ├── caseless@0.12.0
        │ ├─┬ combined-stream@1.0.5
        │ │ └── delayed-stream@1.0.0
        │ ├── extend@3.0.1
        │ ├── forever-agent@0.6.1
        │ ├─┬ form-data@2.1.4
        │ │ └── asynckit@0.4.0
        │ ├─┬ har-validator@4.2.1
        │ │ ├─┬ ajv@4.11.8
        │ │ │ ├── co@4.6.0
        │ │ │ └─┬ json-stable-stringify@1.0.1
        │ │ │   └── jsonify@0.0.0
        │ │ └── har-schema@1.0.5
        │ ├─┬ http-signature@1.1.1
        │ │ ├── assert-plus@0.2.0
        │ │ ├─┬ jsprim@1.4.1
        │ │ │ ├── assert-plus@1.0.0
        │ │ │ ├── extsprintf@1.3.0
        │ │ │ ├── json-schema@0.2.3
        │ │ │ └─┬ verror@1.10.0
        │ │ │   └── assert-plus@1.0.0
        │ │ └─┬ sshpk@1.13.1
        │ │   ├── asn1@0.2.3
        │ │   ├── assert-plus@1.0.0
        │ │   ├── bcrypt-pbkdf@1.0.1
        │ │   ├─┬ dashdash@1.14.1
        │ │   │ └── assert-plus@1.0.0
        │ │   ├── ecc-jsbn@0.1.1
        │ │   ├─┬ getpass@0.1.7
        │ │   │ └── assert-plus@1.0.0
        │ │   ├── jsbn@0.1.1
        │ │   └── tweetnacl@0.14.5
        │ ├── is-typedarray@1.0.0
        │ ├── isstream@0.1.2
        │ ├── json-stringify-safe@5.0.1
        │ ├─┬ mime-types@2.1.17
        │ │ └── mime-db@1.30.0
        │ ├── oauth-sign@0.8.2
        │ ├── performance-now@0.2.0
        │ ├── qs@6.4.0
        │ ├── safe-buffer@5.1.1
        │ ├── stringstream@0.0.5
        │ ├─┬ tough-cookie@2.3.3
        │ │ └── punycode@1.4.1
        │ ├── tunnel-agent@0.6.0
        │ └── uuid@3.1.0
        ├─┬ rimraf@2.6.2
        │ └─┬ glob@7.1.2
        │   ├── fs.realpath@1.0.0
        │   ├── inflight@1.0.6
        │   ├─┬ minimatch@3.0.4
        │   │ └─┬ brace-expansion@1.1.8
        │   │   ├── balanced-match@1.0.0
        │   │   └── concat-map@0.0.1
        │   └── path-is-absolute@1.0.1
        ├── semver@5.4.1
        ├─┬ tar@2.2.1
        │ ├── block-stream@0.0.9
        │ ├─┬ fstream@1.0.11
        │ │ └── graceful-fs@4.1.11
        │ └── inherits@2.0.3
        └─┬ tar-pack@3.4.1
          ├── fstream-ignore@1.0.5
          ├─┬ once@1.4.0
          │ └── wrappy@1.0.2
          ├─┬ readable-stream@2.3.3
          │ ├── core-util-is@1.0.2
          │ ├── isarray@1.0.0
          │ ├── process-nextick-args@1.0.7
          │ ├── string_decoder@1.0.3
          │ └── util-deprecate@1.0.2
          └── uid-number@0.0.6

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@~0.1.4 (node_modules/noble/node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
````

But I'm not sure if this means everything compiled and is ok. How to check if noble installed properly?

Kepete, 2018-01-25:
I have same problem like you, have been trying to find a solution for days. 
I run into exact same problems until I tried `sudo npm install -g --unsafe-perm noble`

[image]

But it seems that it is still not correctly installed. The program that needs it says problem in noble. Any ideas what else can be wrong? 

Before installing noble I used the command `sudo apt-get install libbluetooth-dev libudev-dev`

closed 2018-01-31 by sandeepmistry

yagizyalcintas, 2019-04-23:
May I ask why this issue is closed while everyone (including me) having the same problem while no solution is given?

MMrj9, 2019-05-21:
having the same problem on MAC OS

dsteinman, 2019-05-25:
This ticket should be reopened.  On a fresh Raspbian install noble is still not compiling on RPi3.

```
/home/pi/.node-gyp/10.15.3/include/node/v8config.h:324:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
binding.target.mk:99: recipe for target 'Release/obj.target/binding/src/BluetoothHciSocket.o' failed
make: *** [Release/obj.target/binding/src/BluetoothHciSocket.o] Error 1
make: Leaving directory '/home/pi/node_modules/bluetooth-hci-socket/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:189:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Linux 4.14.98-v7+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/node_modules/bluetooth-hci-socket
gyp ERR! node -v v10.15.3
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/package.json'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/bluetooth-hci-socket):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

+ noble@1.9.1
added 4 packages from 4 contributors and audited 121 packages in 75.95s
found 2 low severity vulnerabilities
  run `npm audit fix` to fix them, or `npm audit` for details
```

The exact commands I ran were:

```
wget https://nodejs.org/dist/v10.15.3/node-v10.15.3-linux-armv7l.tar.gz
tar xfz node-v10.15.3-linux-armv7l.tar.gz
cd node-v10.15.3-linux-armv7l
sudo cp -R * /usr/local/
sudo apt-get install git
sudo apt-get install gcc g++ make
sudo apt-get install libbluetooth-dev libudev-dev
npm install noble
```

using `npm install --unsafe-perm noble` has no effect

using node v12 has no effect