noble/noble #713: error

anand-surkali, 2017-11-22:
[noble.zip](https://github.com/sandeepmistry/noble/files/1496062/noble.zip)
hi,

I am getting an issue "throw new Error('No compatible USB Bluetooth 4.0 device found!');".

My Bluetooth  is  "QCA9377 Bluetooth 4.1". when i add USB VID | USB PID in  usb.js. This is showing new error 

noble
F:\app\bluetooth\bluetooth4\node_modules\usb\usb.js:33
        this.__open()
             ^
Error: LIBUSB_ERROR_NOT_SUPPORTED
    at Error (native)
    at Device.usb.Device.open (F:\app\bluetooth\bluetooth4\node_modules\usb\usb.js:33:7)
    at BluetoothHciSocket.bindUser (F:\app\bluetooth\bluetooth4\node_modules\bluetooth-hci-socket\lib\usb.js:73:19)
    at BluetoothHciSocket.bindRaw (F:\app\bluetooth\bluetooth4\node_modules\bluetooth-hci-socket\lib\usb.js:28:8)
    at Hci.init (F:\app\bluetooth\bluetooth4\lib\hci-socket\hci.js:101:35)
    at NobleBindings.init (F:\app\bluetooth\bluetooth4\lib\hci-socket\bindings.js:82:13)
    at Noble.<anonymous> (F:\app\bluetooth\bluetooth4\lib\noble.js:54:22)
    at emitTwo (events.js:106:13)
    at Noble.emit (events.js:191:7)
    at _addListener (events.js:226:14)
[image]

Thanks in advance

[noble.zip](https://github.com/sandeepmistry/noble/files/1496057/noble.zip)

gpgabriel, 2017-11-28:
The error is self explanatory and is not related to this repository, it should be posted here `https://github.com/tessel/node-usb/issues`. 

https://github.com/tessel/node-usb/blob/60bc9eeab4ae6626e9908803b3c2e98ebfd8df5e/Readme.md#windows might be helpful.

sandeepmistry, 2018-01-31:
@gpgabriel thanks for helping out!

closed 2018-01-31 by sandeepmistry