noble/noble #717: Design change: Pass context objects as parameters ...

nkolban, 2017-12-02:
Howdy folks, new to Noble so please forgive if I am missing something obvious.

When we get a peripheral object, we can invoke its `connect()` method passing a callback function to be invoked when the connect succeeds.  The signature for the callback function is:

```
callback(error)
```

Now imagine I have a function which I write which wants to perform work when connected ... for example

```
function onConnected(error) {
  // do some work
}
```

The problem (to my eyes) is that I am not passed the context of the connection ... to put it clearer, I am not passed a reference to the peripheral.  I would have imagined that the callback should have had the signature:

```
function onConnected(error, peripheral) {
  // do some work with the peripheral
}
```

While I understand that I can register a function as a lambda to be called back and it will receive the contextual peripheral, not being passed the context limits my coding/design choices.

casper288, 2017-12-06:
You can do : 
`peripheral.connect();
peripheral.once('connect', function() {

});`

Or : 
`peripheral.connect(function(error){

});
`

nkolban, 2017-12-06:
Thank you sir.  While those techniques will indeed allow one to achieve the goal of calling user written code with the context of a peripheral.  The notion in the original post was a suggestion for an enhancement / alternative.  What I seem to find is precedent in other API libraries that deal with connection handles where the context of the connection is passed along to the callback functions.  This allows functions to be written which then become generically re-usable.  The post wasn't to report a bug but instead to ask if there were any other users of this package that might consider the passing of context directly into the callback function a useful proposition.

sandeepmistry, 2018-01-31:
API's are based around the Node core API's like https://nodejs.org/dist/latest-v8.x/docs/api/net.html#net_socket_connect

One option is to use `.bind(...)` on the callback to bind context to the callback.

You might be interested in noble-device as well, for a high level API https://github.com/sandeepmistry/noble-device

closed 2018-01-31 by sandeepmistry