noble/noble #718: How do I "shutdown" noble?

nkolban, 2017-12-02:
When I write a node.js app that uses noble and register some event handlers ... for example:

```
noble.on("stateChange", ...);
noble.on("discover", ...);
```

I find that the app no longer terminates.  This makes sense since there is still "logic" running that is waiting for events.  How can I declare that I am no longer interested in these events with the anticipated result that my program will now cleanly exit?

homedad, 2017-12-03:
In node you can remove listener with emitter.removeListener()

see https://nodejs.org/api/events.html#events_emitter_removelistener_eventname_listener <https://nodejs.org/api/events.html#events_emitter_removelistener_eventname_listener>

nkolban, 2017-12-03:
Thank you sir.  I'll give that a try.  I'll close the issue withe the belief that your recipe will work great.

closed 2017-12-03 by nkolban

AlexDM0, 2018-05-23:
Hi, the removelistener doesnt really shut noble down for me. I searched through your code for the listener to "removeListener" which might kill the process.

There is a "onExit" method in the hci-socket/bindings.js@line101 but that is only called on a sigint or exit events from Process. After the init at https://github.com/noble/noble/blob/master/lib/noble.js#L57 the internal bindings are not cleaned up if the user removesListeners from your API.

Is there a method other than process.exit() to cleanly stop all of nobles processes?

Regards