noble/noble #719: Segmentation fault on Alpine Linux

pattyland, 2017-12-07:
I'm trying to run noble inside docker and node crashed on initialisation:

`bash-4.4# cat test.js 
var noble = require('noble');
bash-4.4# node test.js 
Assertion failed: !uv__is_closing(handle) (src/unix/poll.c: uv_poll_start: 120)
Aborted
bash-4.4# node test.js 
Segmentation fault`

Any ideas? Maybe a missing package?

Greets, Sören

sandeepmistry, 2018-01-31:
Please try again with noble master, and reopen if it still happens.

sandeepmistry, 2018-02-05:
@oroce can you narrow down what's triggering this. Are there any Bluetooth interfaces bridged to the docker instance?

oroce, 2018-02-05:
No interface is bridged yet. If I do `require('noble')` I'm receiving: `Assertion failed: loop->watchers != NULL (src/unix/linux-core.c: uv__platform_invalidate_fd: 142)`. But if I do `setTimeout(() => require('noble'), 1000)` I see `Assertion failed: !uv__is_closing(handle) (src/unix/poll.c: uv_poll_start: 120)`.

Docker output is attached below.

The dockerfile I used: https://github.com/oroce/ble-presence-addon/blob/master/Dockerfile

`docker build`:
```
(master)⚡ % docker build --build-arg BUILD_FROM="homeassistant/amd64-base:latest" -t local/ble-presence-addon:latest .                                       
Sending build context to Docker daemon  3.714MB
Step 1/9 : ARG BUILD_FROM
Step 2/9 : FROM $BUILD_FROM
 ---> b7f4cb286eff
Step 3/9 : ENV LANG C.UTF-8
 ---> Using cache
 ---> ea2c2f7520e5
Step 4/9 : RUN apk add --no-cache jq nodejs nodejs-npm python build-base
 ---> Using cache
 ---> e9cb11256627
Step 5/9 : RUN npm set unsafe-perm true
 ---> Using cache
 ---> 0649d44686aa
Step 6/9 : WORKDIR /usr/src/app
 ---> Using cache
 ---> ade7487263ae
Step 7/9 : COPY . .
 ---> de882dce0b73
Step 8/9 : RUN chmod a+x ./run.sh
 ---> Running in 27cd0f81f02d
Removing intermediate container 27cd0f81f02d
 ---> 9d0704af5caa
Step 9/9 : CMD [ "./run.sh" ]
 ---> Running in 57d5a4cd2a57
Removing intermediate container 57d5a4cd2a57
 ---> d959dece3276
Successfully built d959dece3276
Successfully tagged local/ble-presence-addon:latest
```

`docker run`:
```
(master)⚡ % docker run --rm -it local/ble-presence-addon:latest                                                                                               ~/developing/home-assistant/ble-tracker-addon
starting version 3.2.4

> usb@1.3.1 install /usr/src/app/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/tessel/node-usb/releases/download/1.3.1/usb_bindings-v1.3.1-node-v57-linux-x64.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for usb@1.3.1 and node@8.9.3 (node-v57 ABI, musl) (falling back to source compile with node-gyp)
make: Entering directory '/usr/src/app/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
In file included from ../libusb/libusb/os/linux_usbfs.c:43:0:
../libusb/libusb/os/linux_usbfs.h:24:25: fatal error: linux/types.h: No such file or directory
 #include <linux/types.h>
                         ^
compilation terminated.
make: *** [libusb.target.mk:134: Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o] Error 1
make: Leaving directory '/usr/src/app/node_modules/usb/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at emitTwo (events.js:126:13)
[... 29 lines cut ...]
../src/BluetoothHciSocket.cpp:275:72: warning: 'v8::Local<v8::Object> v8::Function::NewInstance(int, v8::Local<v8::Value>*) const' is deprecated: Use maybe version [-Wdeprecated-declarations]
   Local<Value> error = errorConstructor->NewInstance(1, constructorArgs);
                                                                        ^
In file included from /root/.node-gyp/8.9.3/include/node/v8.h:26:0,
                 from /root/.node-gyp/8.9.3/include/node/node.h:63,
                 from /root/.node-gyp/8.9.3/include/node/node_buffer.h:25,
                 from ../src/BluetoothHciSocket.cpp:7:
/root/.node-gyp/8.9.3/include/node/v8.h:3832:31: note: declared here
                 Local<Object> NewInstance(int argc, Local<Value> argv[]) const);
                               ^
/root/.node-gyp/8.9.3/include/node/v8config.h:318:3: note: in definition of macro 'V8_DEPRECATED'
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
  SOLINK_MODULE(target) Release/obj.target/binding.node
  COPY Release/binding.node
make: Leaving directory '/usr/src/app/node_modules/bluetooth-hci-socket/build'
npm WARN ble-tracker-addon@1.0.0 No repository field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.3.1 (node_modules/usb):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.3.1 install: `node-pre-gyp install --fallback-to-build`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

added 91 packages in 17.962s
Assertion failed: loop->watchers != NULL (src/unix/linux-core.c: uv__platform_invalidate_fd: 142)
Aborted
```

kuzyn, 2018-05-16:
@oroce I'm seeing the same error except that in my case it works when built with Docker but does not when built through Docker Compose. The compose file is simply a named service with build context and env variable.

The command I used with docker is
```
docker run --network host --restart unless-stopped -t my-image-name:latest
```

Edit: I just realized that I get the same effect in Docker if I run without the `--network host`

@sandeepmistry Do you have any pointer as to how to debug this? I can root into the failing container & I'm able to sink in a couple of hours to see this through

UsersDoProgram, 2018-06-06:
@kuzyn When I add --network host to my docker run statement output goes from, 'Segmentation fault' to 'docker: Error response from daemon: OCI runtime create failed: container_linux.go:348: starting container process caused "exec: \"--network\": executable file not found in $PATH": unknown.'