noble/noble #721: Can't install with -g: xpc-connection mkdir error

llemtt, 2017-12-20:
Hi, trying to install globally with -g option doesn't work, xpc-connection fails with a "mkdir" error, everything works without -g!

-------------
TeoPreviati:~ teo$ sudo npm install noble -g

> xpc-connection@0.1.4 install /usr/local/lib/node_modules/noble/node_modules/xpc-connection
> node-gyp rebuild

gyp ERR! configure error
gyp ERR! stack Error: EACCES: permission denied, mkdir '/usr/local/lib/node_modules/noble/node_modules/xpc-connection/build'
gyp ERR! stack     at Error (native)
gyp ERR! System Darwin 14.5.0
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /usr/local/lib/node_modules/noble/node_modules/xpc-connection
gyp ERR! node -v v6.2.2
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/noble/node_modules/bluetooth-hci-socket):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for bluetooth-hci-socket@0.5.1: wanted {"os":"linux,android,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/noble/node_modules/xpc-connection):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

+ noble@1.8.1
updated 1 package in 11.134s

llemtt, 2017-12-21:
Updated node to 9.2.1, probelm solved!

closed 2017-12-21 by llemtt

playaspec, 2018-10-09:
Upgrading to 9.x isn't an option for people tied to 8.x