noble/noble #724: descriptors got from discoverDescriptors() only contain 16-bit uuid even for custom descriptors

sfyang, 2017-12-21:
Hello,

While writing a script, running on Ubuntu Linux 16.04 with BlueZ 5.37, for finding all the descriptors of a custom characteristic, the uuid of the descriptor objects passed to the callback by discoverDescriptors() of the characteristic contain only 16-bit value. However, since the characteristic contains at least one custom descriptors, the 16-bit value can not be used to  reliably identify the custom descriptor by matching the full 128-bit UUID. Judging from the value I got, I think currently the truncated 16-bit value of the custom descriptor is only the less significant 2-byte of the full 16-byte value.

I did a little code digging, and, since I was running the script on Linux which uses the "hci-socket" binding, it looks like that the problem is caused by the way the "Find Info Response" was parsed between line 668 and 677 in  gatt.js. Is there way to get the full 128-bit UUID for custom descriptors?

AlexPeitz, 2017-12-21:
Yes you are right. That is because every time a find info request using characteristic handles are sent, the peripheral replies in batches with either 16 bit or 128 bit descriptor UUIDs, with a header containing as first byte the opcode and as second byte an identifier saying if the following payload contains handle-uuid pairs of 16 bit uuids or 128 bit uuids. The line `num = data[1]` reads that identifier, but does not take into account if it is a 128 bit value.

I fixed this by replacing the for loop in line 671 : 

```
for (i = 0; i < num; i++) {
        descriptors.push({
          handle: data.readUInt16LE(2 + i * 4 + 0),
          uuid: data.readUInt16LE(2 + i * 4 + 2).toString(16)
        });
}
```

with the following code:

```
var format = data[1];
      switch(format){
        case 1:
          var num = (data.length - 2) / 4;
          for (var i = 0; i < num; i++) {
            descriptors.push({
              handle: data.readUInt16LE(2 + i * 4 + 0),
              uuid: data.readUInt16LE(2 + i * 4 + 2).toString(16)
            });
          }
        break;
        case 2:
          var num = (data.length - 2) / 18;
          data = data.slice(2);
          for (var i = 0; i < num; i++) {
            var h = data.readUInt16LE(i * 18);
            var uuidBuf = data.slice(2 + i*18, (i+1)*18);
            var u = "";
            for (var j = 0; j < uuidBuf.length; j++) {
              var newChar = uuidBuf.readUInt8(j).toString(16);
              if (newChar.length < 2)
                newChar = "0" + newChar;
              u = newChar + u;
            }
            descriptors.push({
              handle: h,
              uuid: u
            });
          }
        break;
      }
```

I replaced `num` with `format` because I thought it is easier to understand that way.

sfyang, 2017-12-22:
@AlexPeitz Thanks for the explanation and the code snippet. However, the line that slice the 128-bit uuid from the buffer, `var uuidBuf = data.slice(2 + i*18, (i+1)*18);`, seems to have an incorrect end index. Doesn't the end index be `2 + (i+1)*18` rather than just `(i+1)*18` since the `2` counts for the 2-byte handle field?

AlexPeitz, 2017-12-22:
@sfyang glad I could help. The end Index is correct that way, because I iterate over handle-uuid pairs which are 18 bytes long. In these 18 bytes, the first 2 bytes are the 16 bit handle, the following 16 bytes are the 128 bit uuids. My variable `uuidBuf` is representing always one UUID at a time, so I have to slice the buffer starting at the 3rd byte of each handle-uuid pair and until the end of that pair, which is `(i+1)*18`. So yes, basically the 2 handle bytes are contained in the factor 18.

sfyang, 2017-12-23:
@AlexPeitz Yes, you are right. My eyes saw 18, but my brain ignored it and thought it was 16. Sorry for my stupid comment. Thanks.

So what should I do now? Should I try to create a pull request for this issue?

AlexPeitz, 2017-12-23:
Yes that's a good idea, you could do that! I'm a complete github noob so I've got no idea how to submit a PR 😅 So if you know how, then you could do that, otherwise I'll have to learn!

sfyang, 2017-12-23:
Well, I haven't submit any pull request on GitHub so far :stuck_out_tongue_closed_eyes: Most of the time if I can find any possible fix or workaround for the issue I found, I attached the patch(es) to the issue report for developers' reference. So I did have read some version of the "[Create a pull request](https://help.github.com/articles/creating-a-pull-request/)" article by GitHub, I haven't had any experience on that 😅
If this does not cost you too much of your time, would you please do that for this issue? Thanks in advance.

AlexPeitz, 2017-12-23:
Okay I added a PR, let's hope that I did it the right way!