noble/noble #728: State = unknown on BLE supported mac

virtapoika, 2018-01-05:
Hi, 

I'm getting started with noble and I found a problem with my simple code

`var noble = require('noble');

console.log(noble.state);
noble.on('stateChange', function(state) {
    if (state === 'poweredOn')
    {
        console.log("ebin")
        noble.startScanning();
    }
    else
        noble.stopScanning();
});

noble.on('discover', function(peripheral) {
    console.log('Found device with local name: ' + peripheral.advertisement.localName);
    console.log('advertising the following service uuid\'s: ' + peripheral.advertisement.serviceUuids);
    console.log();
});`

The state never changes "powerOn" and the state is "unknown". I'm running on Macbook pro 2017 so BLE is supported.

Help needed!

webartistse, 2018-01-11:
Try noble.state = 'poweredOn' on last row. This will change the state and trigger the events.

mm-ns, 2018-01-15:
where exactly? I tried to set noble.state = 'poweredOn'  immediately as first thing but didn't help and neither at the end.

petewall, 2018-01-15:
I also am running the same program at @virtapoika and the stateChange handler is never invoked.  Setting the `.state` variable directly doesn't seem to change anything.

On a Mid 2015 MacBookPro, running Mac OS X 10.13.2

mm-ns, 2018-01-15:
I'm also on High Sierra and had it working prior....

mm-ns, 2018-01-15:
I tried the same deleting all bluetooth plist but no luck..

webartistse, 2018-01-15:
Weird, my code that worked yesterday for scanning has stoped working. Only thing I've done is update OS from Sierra to High Sierra.

mm-ns, 2018-01-15:
here is a solution --> https://github.com/sandeepmistry/noble/issues/679

just add this to your package.json : "noble": "git://github.com/jacobrosenthal/noble.git#highsierra",

instead of noble. works for now. :)

petewall, 2018-01-15:
Thanks, @mm-ns.  That worked for me, too.

sandeepmistry, 2018-01-31:
Closed via https://github.com/sandeepmistry/noble/pull/727.

closed 2018-01-31 by sandeepmistry