noble/noble #729: Calling discoverCharacteristics for a specific UUID list forgets previously discovered characteristic uuids

elafargue, 2018-01-05:
I scratched my head on this one for a long time: I have a BLE device that shows a very common behaviour: you write on one characteristic, you receive data on another.

In my code, I first do a service/characteristic discovery to subscribe to the "read" characteristic, then whenever I want to write, I also go through the same exercise: service/characteristic discovery to retrieve the 'write' characteristic, then write to it. This way, I never query characteristics I don't need ahead of time, I just discover as I go along.

Unfortunately, on the Mac, if you look at yosemite.js , the list of characteristics for a service is reset each time you call characteristic discovery. This means that you have to either discover every single UUID you will ever need in every discoverCharacteristics call, or you will lose any UUID you discovered previously but didn't include in the latest discovery.

An easy fix is to do something like...

```
if (! this._peripherals[deviceUuid].services[serviceStartHandle].characteristics)
    this._peripherals[deviceUuid].services[serviceStartHandle].characteristics = {};
```

...in yosemite.js. This avoids forgetting any already known characteristic. This solves my issue, but there might be side effects which I have not identified yet. I assume other OS implementations have the same issue.

Alternatively, I suppose it's possible to structure code to discover services and characteristics only once per connection, or to make sure you always discover all the characteristics you will ever need, but this should be properly documented...

elafargue, 2018-01-06:
Note: the current behaviour leads for instance to crashes when unsubscribing to a characteristic that was lost in a subsequent call to discoverCharacteristics, and also leads to not being able to receive data from a previously subscribed-to characteristic in case a call to discoverCharacteristics is made after subscription.

elafargue, 2018-01-06:
Created PR #730  which implements the fix.

jrobeson, 2018-08-07:
@elafargue : so can this be closed then?

elafargue, 2018-08-07:
Yes, PR #747 fixed this.

closed 2018-08-07 by elafargue