noble/noble #733: Error: data must be a Buffer, when writeing to characteristics using webpack and electron-builder

elovin, 2018-01-09:
When I try  to write data to a characteristic I get "Error: data must be a Buffer" but it definitely is a Buffer
and it works when I run a test using node-ts.

Is it somehow possible to use another type ? 

```
  protected writeNextChunkToPeripherial(): Promise<boolean> {

    return new Promise((resolve, reject) => {

      let nextChunk: Buffer = super.getNumberOfChunksFromResponseContainer(20);

      if (nextChunk.length == 0) return resolve(true);
      super.getCurrentPeripherialMainServiceCharacteristics().write(nextChunk, true, (error) => {
        if (error) {
          reject(error);
        }
        else {
          resolve(false);
        }
      });

    });
  }
```

sandeepmistry, 2018-02-12:
[Buffer.isBuffer(...)](https://nodejs.org/dist/latest-v8.x/docs/api/buffer.html#buffer_class_method_buffer_isbuffer_obj) should be used in [characteristic.write(...)](https://github.com/noble/noble/blob/master/lib/characteristic.js#L62-L66) instead of the current "instanceof".

@elovin do you have time to try this out and submit a pull request if that works.

elovin, 2018-03-12:
sorry for the late reply, I will try this

kunakasaki, 2018-12-25:
> @elovin @sandeepmistry i am trying to use nobel with electronjs build. i am keep getting this warning message "adapter state unauthorized, please run as root or with sudo"
> If you know a way to authorize the adapter. please let me.
> Code is running perfectly with direct node command but unable to run inside electron app.
> sudo setcap cap_net_raw+eip $(eval readlink -f `which node`

same here, ran out of options

closed 2020-01-23 by elovin