noble/noble #734: Almost all found devices have localName undefiend

webartistse, 2018-01-11:
I did a little test script with following code:

`var noble = require('noble');

noble.on('stateChange', function(state){
    if(state == 'poweredOn') {
        console.log('Bluetooth is on.');
        noble.startScanning();
    }
});

noble.on('scanStart', function(){
    console.log("start scanning...");
});

noble.on('discover', function(peripheral){
    console.log(peripheral.advertisement.localName);
});

noble.state = 'poweredOn';`

Almost every device has localName undefined. I only see the name of three devices. These three is Android mobiles. I guess the undefined devices is Apple products I have around like mobiles, tablets, computers, mouse etc. 

What it wrong? Im running the script on MacOS Sierra and Node 8.4.

webartistse, 2018-01-12:
Tried same code on Raspbian with Node 9 at home. I can see my Samsung TV with localName but rest is undefined. It's probably my iPhone, iPad, Magic Mouse and Macbook.

fcoury, 2018-01-12:
Same issue here.

webartistse, 2018-01-12:
I tried advertisement-discovery.js in example folder for Noble and it looks like this. Only see two Android phones. Its alot of iPhones, iPads and Macs around here and they are undefined.

`peripheral discovered (acbf5e2f71f7414ca11bfd72dbee3209 with address <unknown, unknown>, connectable true, RSSI -66:
	hello my local name is:
		undefined
	can I interest you in any of the following advertised services:
		[]
	here is my manufacturer data:
		"4c00090603010a21190e"

peripheral discovered (851d30a8671f4c8db1392c9a8cb557e3 with address <unknown, unknown>, connectable true, RSSI -85:
	hello my local name is:
		undefined
	can I interest you in any of the following advertised services:
		[]
	here is my manufacturer data:
		"4c000c0e00d824a09c28f58aaa152ee9bcfa"

peripheral discovered (6fbf277c7d7c44a5ba43625f7fa8ac90 with address <unknown, unknown>, connectable true, RSSI -82:
	hello my local name is:
		undefined
	can I interest you in any of the following advertised services:
		[]
	here is my manufacturer data:
		"4c00100501909bb558"

peripheral discovered (8325a32605aa4ffe852d1277e700a229 with address <unknown, unknown>, connectable true, RSSI -84:
	hello my local name is:
		undefined
	can I interest you in any of the following advertised services:
		[]
	here is my manufacturer data:
		"4c0010050a10048563"

peripheral discovered (42b52cacae70405e9da4fad7bbe7f717 with address <unknown, unknown>, connectable true, RSSI -92:
	hello my local name is:
		HTC BS 873FF0
	can I interest you in any of the following advertised services:
		["cb00"]

peripheral discovered (ac4d9831f86e4a3baefcebdf18c1d916 with address <unknown, unknown>, connectable true, RSSI -83:
	hello my local name is:
		HTC BS E22E97
	can I interest you in any of the following advertised services:
		["cb00"]`

cmvee, 2018-01-15:
Same issue. Running Node 8.9.4. Tested on both Raspbian Stretch (BlueZ 5.48) and on Mac OS 10.12.6.

Scan results on Mac OS consistently shoe the localName properly populated. The same code running on Linux (under the same node 8.9.4) shows name undefined.

On Linux, running `hcitool name <bluetooth MAC address>` shows the name of many of the devices that are showing up as unnamed via noble.

webartistse, 2018-01-15:
I think I got the answer. A friend told me that Apple devices does not send this data for integrity reasons. If you turn off and turn on bluetooth you will also get a new ID and address for your device. He told me that I can find this in Apple developer documentation but havn't looked it up yet.

cmvee, 2018-01-16:
@webartistse thanks for chiming in, but everything works fine on Mac OS. It's Raspbian for me that has the issue.