noble/noble #737: Cannot assign peripheral object to a variable

bjornspockeli, 2018-01-15:
Hi, 
I have 3 peripheral nodes that I want to connect to my Raspberry Pi 3. I have managed to connect to all the peripherals by placing all my code nested within the noble.on('discover', function(peripheral) function, which is rather messy.

What I would like to do is to set the peripheral object passed in the discovery callback to a variable, e.g. PeripheralNode and reference this instead of the peripheral object, like shown below. However, I get the following error

```
/home/pi/code/node/noble_experimental/noble_app.js:61
OfficeNode.once('connect', function(){
          ^

TypeError: Cannot read property 'once' of null
```
In Sandeep's pizza example he assigns the characteristics to vairables, so I do not quite understand why the approach below does not work. Is this possible? If not why not?

I am a Node.js and Noble novice so bear with me.
```
var noble = require('noble');

var PeripheralNode = null;

// Custom UUIDs
var customServiceUUID 					= 'f364140000b04240ba5005ca45bf8abc';

// State Change Callback
noble.on('stateChange', function(state) {
	switch(state){
		case 'unknown':
			console.log('State: Unknown');
			break;
		case 'resetting':
			console.log('State: Resetting');
			break;
		case 'unsupported':
			console.log('State: Unsupported');
			break;
		case 'unauthorized':
			console.log('State: Unauthorized');
			break;
		case 'poweredOn':
			console.log('State: PoweredOn');
			console.log('Starting Scanning...');
			noble.startScanning([customServiceUUID]); // any service UUID, no duplicates 
			break;
		case 'poweredOff':
			console.log('State: PoweredOff');
			// Do nothing
			break;
		default:
			noble.stopScanning();
			break;
	}
});

// Peripheral Discovery Callback
noble.on('discover', function(peripheral) {
	console.log('\n');
	console.log('Found Peripheral with address:', peripheral.address);
		
	noble.stopScanning();

    PeripheralNode = peripheral;

	PeripheralNode.connect( function(err) {
		if(err) {
			console.log('Connect Error:', err);	
		}
	})
});

// Connect Callback
PeripheralNode.once('connect', function(){   
    console.log('Connected to peripheral:', OfficeNode.localName);
});
```

jguillod, 2018-01-17:
You should put`PeripheralNode.once('connect', function(){ ... });` just after the statement `PeripheralNode = peripheral;` !
NodeJS is async, remember?
;o)

bjornspockeli, 2018-01-17:
I know its asynchronous , but what I wanted was to avoid putting a callback within a callback to keep the code cleaner. In my head there should not be any difference with declaring the callback as shown above and right after the `PeripheralNode = peripheral;` statement.

sandeepmistry, 2018-01-31:
Add a console.log you'll find the answer ;)

closed 2018-01-31 by sandeepmistry