noble/noble #740: Can't install noble correctly

Kepete, 2018-01-24:
I need noble for a plugin, but can't figure out how to install it correctly. 
I have tried command `sudo npm install noble` and it didn't install it at all. Then I tried it using root user and got this output: 

[image]

the plugin what needs it runs into an error and this is because of noble is installed incorrectly.

Kepete, 2018-01-25:
I also tried the command `sudo npm install -g --unsafe-perm noble`
It kind of worked, but my plugin what needs noble still says that it is incorrectly installed. 

[image]

M1ras, 2018-01-29:
I've run into this issue as well, it looks like the USB bindings don't play nice with Node 9.X. Try reinstalling with Node 8.X LTS (it's what fixed it for me).

sandeepmistry, 2018-01-31:
It's an optional dependency on Linux so this is fine, `node-bluetooth-hci-socket` has 1.x of node-usb in package.json and 1.4.0 is the latest release. Will have to wait for upstream to fix node 9 stuff.

closed 2018-01-31 by sandeepmistry