noble/noble #741: What does noble@2.0 look like?

jacobrosenthal, 2018-01-28:
@sandeepmistry and I are summiting at iotdevfest this weekend and are (at the very early stage) of pondering what a 2.0 looks like?

Bindings layers updates an obvious change:

OSX
Get rid of xpcids in darwin as these break every year or two. Though frankly these breaking is the only time anyone cares or returns to the project. Sadly being stable in open source is a death sentence (more below)

Sandeeps been working on a lower level objc
https://github.com/sandeepmistry/node-core-bluetooth ffi but we found it was way too slow and leaks memory
https://github.com/sandeepmistry/node-core-bluetooth/tree/node-objective-c-runtime fast but leaks memory
https://github.com/sandeepmistry/noble/tree/core-bluetooth

Rewriting with proper memory management should clear that up and make that route viable

WINDOWS
The windows people did some good uwp that needs more help to make nobley. Could we get Microsoft involved? Maybe they could sponsor the 2.0 branch and we can bring this in proper with a big release?
https://github.com/sandeepmistry/noble/issues/550

LINUX
We've heard there might be problems on linux but what though? Which tickets? Or open new ones please. dbus/bluez instead hci has been pushed by those guys.
https://github.com/sandeepmistry/noble/issues/310
need more info though, is new enough dbus available on all platforms like rapsbian, etc???
Need some linux leaders, can we get the bluez people to get in here with actual code vs just issues?

Bindings abstraction and tests
(and low level errors bubbling up finally)
https://github.com/sandeepmistry/noble/pull/431

Streams (we get these for 'free' by dumping older node versions finally)
https://github.com/sandeepmistry/noble/pull/593

Promises
https://github.com/sandeepmistry/noble/pull/708

Other API changes? Whats your wishlist?

Support and attention
Were constantly used in commercial work and noble is, I would argue, incredibly stable so we sadly only hear from those people when xpcids break and they want to 'help', by asking when it will be fixed. We have an open collective like everyone else but that doesnt mean much without some explicit push and plan for use of those funds. How do we use the 2.0 release get more community and issue triage and yes, funding in order to reward such by say sending hardware and stickers and shirts to contribs, etc. Ive always wanted a mailbag section with new ble hardware and tutorials as well. Companies would gladly send toys to contribs to make and test noble support and to get their hardware seen...

@monteslu @noopkat @jrobeson @explicitcall and please @ anyone else who might have comment here.

grover, 2018-01-28:
Hi @jacobrosenthal,

my 2 cents on this: You've done a great job with this library and the support it gives to other node projects. I myself used this to develop homebridge-ranger, a homebridge plugin that acts as a HomeKit extender to otherwise badly reachable BLE HomeKit accessories.

From my experience the library is difficult to use in two scenarios: On OS X the reliance on HCI and circumventing CoreBluetooth proves difficult in the discovery process of BLE devices. The discovery process of the (otherwise hidden) hardware address is broken in the sense that unless you've connected to the device, you won't have an address. And even if you did connect, the device will not be updated accordingly after CoreBluetooth did assign an address.

The other scenario is on Raspbian, where there's multiple issues related to disconnects as already documented in other issues. For ranger my choice was to stop the discovery process while trying to connect until the connection has been established. Also dealing with multiple (concurrent) connections has seen similar difficulties with random disconnects or connection failures. This has proven to be a time consuming solution as it did require some thought on serialization of discovery, connections, reads and writes. (Side note: HomeKit requires continuous discovery as accessories signal events in the manufacturer data, when not actively connected.)

These two issues notwithstanding this has been a great library and certainly only this gave me the ability to create the ranger as quickly as I did.

Thank you very much!

M1ras, 2018-01-29:
Hi @jacobrosenthal, sorry I missed you and @sandeepmistry at IoT Devfest. I think it'd be nice if Noble...
  - better documented Object properties[₁](https://github.com/sandeepmistry/noble#peripheral-discovered).
  - exported commonly used Strings like event names. Something along the lines of `noble.on(noble.events.scanStart, () => {});` instead of `noble.on('scanStart', () => {});`. This would help cut down on typos and take better advantage of linters.
  - unify the way unknown values are represented. `peripheral.address` is assigned `'unknown'` if not known[₁](https://github.com/sandeepmistry/noble#peripheral-discovered) while other values are assigned `undefined` if not known.

In our talk earlier, you mentioned Noble doesn't have a whole bunch of active contributors. I'd be more than happy to work on the Linux side of things, particularly dbus/bluez. Thanks to the Noble devs for keeping this project alive.

1: [https://github.com/sandeepmistry/noble#peripheral-discovered](https://github.com/sandeepmistry/noble#peripheral-discovered)

monteslu, 2018-01-29:
Thinking we could change the `noble.startScanning(...)` a bit to line up with the options available to web bluetooth's `requestDevice()` spec.  

We could do a bit of parameter juggling and keep it backwards compatible since the noble API currently expects an array and the webbluetooth one expects an object.  However, if we're doing a major version bump, I don't know if that's necessary.

jrobeson, 2018-01-29:
I wouldn't mind going back to trying to make the dbus API work again.  The API got better in the meantime.  I'd like to have some help though.

So what node version would be the minimum?  4.x is already almost out of support at this point.

jacobrosenthal, 2018-01-31:
https://github.com/nodejs/Release
Especially if we think about supporting async/await and promises at some level I could see us shipping for es6 support (node 7.6), so launch for node 8, maybe we could be in 'beta' for a year until 6x ends support in April 2019 @jrobeson

elafargue, 2018-01-31:
Noble is working extremely well for me on both MacOS and Raspberry Pi - I have devices in production that are running 24/7 and behaving really well.

I had to fix one big issue (PR #730 ) that leads to bad crashes in case discovery is run multiple times, which is a common use case when you run an app for multiple days and regularly connect/disconnect from devices. Apart from this one that required fixing the Noble code, I find that one single codebase works really well on both MacOS and Linux, at least for the app I wrote...

jacobrosenthal, 2018-02-05:
Another out there idea, luis wants more web ble like scanning options. We want to be more promise async/await style which means api changes. What about dropping the noble api and mirroring the web ble as thats the standard now, adding to it where we have to but reacting to it as it adds features.

M1ras, 2018-02-16:
I think mirroring the [Web Bluetooth API](https://webbluetoothcg.github.io/web-bluetooth/) is a good idea. It would allow developers to port Web Bluetooth dependent applications to Node easily.

**IE** A developer could split out the bluetooth dependent part of their code into a separate library, and simply use `navigator.bluetooth` or `require("noble")` depending on the platform they're deploying to.

It looks like Chrome requires a user to choose which Bluetooth device to pair using a native popup, and [node-web-bluetooth](https://github.com/IjzerenHein/node-web-bluetooth) uses a command line prompt. I think we should avoid any sort of UI and allow developers to implement their own prompt **IE** define some sort of interface or function that receives a list of BLE devices and returns one to connect to. There are different Node UI libraries ([Inquirer.js](https://github.com/SBoudrias/Inquirer.js) and [blessed](https://github.com/chjj/blessed) to name a few) and each use their own drawing algorithms, so allowing developers to use their own implementation would avoid a lot of headaches (we could still provide our own canonical implementation in a separate repository).

mane87, 2018-02-22:
I would love to help with the macOS part. 
All though noble is currently stable, i think moving away from xpcids would be beneficial in the long run.

jrobeson, 2018-06-05:
@jacobrosenthal : I was planning on attempting the dbus apis again, but I really don't want to deal with node without promises.  I'll try to come back to it once you folks decide to require node 6.x at the least.

jrobeson, 2018-06-05:
I just looked and debian buster (which includes versions of node > 4.x) isn't due to be released until mid 2019.  I guess i'm not sure how to proceed now.

jrobeson, 2018-06-05:
Oh, i take that back. I thought promises appeared in node 6.x, but it's really 4.x

M1ras, 2018-10-31:
#839 I think this is a good feature to implement in noble 2.0  
I've written [an application](https://gitlab.com/Miras/LSP) that uses noble, and noble arbitrarily exiting instead of bubbling an error would prevent resources from being freed.