noble/noble #748: Why am I getting different UUID when I run in App from running it directly in noble/example/advertisement.js?

NaseebullahSafi, 2018-02-01:
In my App, I'm trying to connect to an arduino using react-native-ble.

I've followed this [`step`](https://github.com/sandeepmistry/noble#windows)

When I run it in noble/example/advertisement.js, it returns the following:
[image]

But when I try to search it in my App, I get the following:

[image]

This is a big issue because I cant connect to my arduino.

My services UUID should be 4 digits long for example:

Should be
=
```
1800
1801
2220
```
Not:
=
```
0000180000001....
0000180100001....
0000222100001....
```

Why is this happening?

NaseebullahSafi, 2018-02-02:
Any ideas? @sandeepmistry

sandeepmistry, 2018-02-02:
Please don't open multiple issues. https://github.com/jacobrosenthal/react-native-ble/issues/58.

closed 2018-02-02 by sandeepmistry