noble/noble #753: noble

srikarvijay4, 2018-02-08:
module.js:460
  return process.dlopen(module, path._makeLong(filename));
                 ^

Error: Module version mismatch. Expected 46, got 59.
    at Error (native)
    at Object.Module._extensions..node (module.js:460:18)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:311:12)
    at Module.require (module.js:366:17)
    at require (module.js:385:17)
    at Object.<anonymous> (/home/pi/node_modules/bluetooth-hci-socket/lib/native.js:3:15)
    at Module._compile (module.js:435:26)
    at Object.Module._extensions..js (module.js:442:10)
    at Module.load (module.js:356:32)

sandeepmistry, 2018-02-11:
Please re-install noble or re-build the native add-on for the latest version of Node.js you are using.

closed 2018-02-11 by sandeepmistry

ocaduacelab, 2018-11-23:
I'm also having this issue, but I don't understand what "re-build the native add-on for the latest version of Node.js you are using" means? Can someone elaborate? Thanks.