noble/noble #754: Manufacturer data truncated on RPi3

iotcara, 2018-02-09:
Hi - apologies in advance if this is a duplicate.

Advertisement-discovery.js returns a shortened version of my expected advertisement's manufacturer data. I noticed the different when I enabled DEBUG on the console:

Debug info (which is what I expect to see):

`gap advertisement ** = {"manufacturerData":{"type":"Buffer","data":[170,187,0,11,87,126,242,121,151,108,2,94,220,111,225,93,183,4,100,0,0,108,246,148,1,31]},"serviceData":[],"serviceUuids":[],"solicitationServiceUuids":[],"serviceSolicitationUuids":[]} +2ms
`

what advertisement-discovery.js returns:
`here is my manufacturer data:
		"aabb6cf69401"`

How do I obtain the correct manufacturer's data through noble in this instance?

Below is a more complete debug extract
`hci onSocketData: 043e2b0201000079f27e570b001f0201061bffaabb000b577ef279976c025edc6fe15db7046400006cf694011fc4 +1s
  hci 	event type = 4 +1ms
  hci 	sub event type = 62 +0ms
  hci 		LE meta event type = 2 +0ms
  hci 		LE meta event status = 1 +0ms
  hci 		LE meta event data = 000079f27e570b001f0201061bffaabb000b577ef279976c025edc6fe15db7046400006cf694011fc4 +0ms
  hci 			type = 0 +1ms
  hci 			address = 00:0b:57:7e:f2:79 +0ms
  hci 			address type = public +0ms
  hci 			eir = 0201061bffaabb000b577ef279976c025edc6fe15db7046400006cf694011f +0ms
  hci 			rssi = -60 +0ms
  gap advertisement ** = {"manufacturerData":{"type":"Buffer","data":[170,187,0,11,87,126,242,121,151,108,2,94,220,111,225,93,183,4,100,0,0,108,246,148,1,31]},"serviceData":[],"serviceUuids":[],"solicitationServiceUuids":[],"serviceSolicitationUuids":[]} +2ms
  hci onSocketData: 043e2b0201040079f27e570b001f07ffaabb6cf6940115096d6d79206c6f676765720000000000000000000000c3 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 62 +0ms
  hci 		LE meta event type = 2 +0ms
  hci 		LE meta event status = 1 +0ms
  hci 		LE meta event data = 040079f27e570b001f07ffaabb6cf6940115096d6d79206c6f676765720000000000000000000000c3 +1ms
  hci 			type = 4 +0ms
  hci 			address = 00:0b:57:7e:f2:79 +0ms
  hci 			address type = public +0ms
  hci 			eir = 07ffaabb6cf6940115096d6d79206c6f676765720000000000000000000000 +0ms
  hci 			rssi = -61 +0ms
  gap advertisement ** = {"localName":"mmy logger\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","manufacturerData":{"type":"Buffer","data":[170,187,108,246,148,1]},"serviceData":[],"serviceUuids":[],"solicitationServiceUuids":[],"serviceSolicitationUuids":[]} +1ms
peripheral discovered (000b577ef279 with address <00:0b:57:7e:f2:79, public>, connectable true, RSSI -61:
	hello my local name is:
		mmy logger
	can I interest you in any of the following advertised services:
		[]
	here is my manufacturer data:
		"aabb6cf69401"`

iotcara, 2018-02-12:
It would be much appreciated if someone can provide some insight/advice/solution to this ...

don, 2018-02-16:
In the output above, your peripheral 00:0b:57:7e:f2:79 lists the manufacturer data as [170,187,108,246,148,1] in the advertisement and "aabb6cf69401" in the output. This is the same data displayed in 2 different formats. This is working as expected.

    $ node
    > b = new Buffer([170,187,108,246,148,1])
    <Buffer aa bb 6c f6 94 01>
    > 

The other buffer contains different manufacturer data, so I'm guessing that the *scan* and *scan response* are returning different manufacturer data. Closing as a duplicate of https://github.com/noble/noble/issues/305.

closed 2018-02-16 by don