noble/noble #755: Multiple interfaces problem

FROeHlyEisvogel, 2018-02-09:
Hey there,
I habe two bluetooth interfaces on my raspberry pi.
One should be used by noble and the other for continuous scanning.
So the problem is that it seems to be impossible to prevent noble touching the second interface.
It seems like noble is resetting all interfaces when loaded and than also performing commands on the second interface even if the first one is selected.

Greetings
Samuel

don, 2018-02-11:
Set NOBLE_HCI_DEVICE_ID to control which adapter noble uses

https://github.com/sandeepmistry/noble#multiple-adapters

FROeHlyEisvogel, 2018-02-15:
Could it be that there is a problem when using this in coffee script?
I did notice now that when I am using the command in a javascript file it does work but not in coffe script.

I am using this command:
process.env['NOBLE_HCI_DEVICE_ID'] = 1

Is there anything I have to change from javascript to coffeescript?

don, 2018-02-16:
> Is there anything I have to change from javascript to coffeescript?

Probably, especially since it works in JavaScript. I don't use coffeescript so I can't help, sorry.

Another option is to set the env from your shell before calling your script

    sudo NOBLE_HCI_DEVICE_ID=1 node <your file>.js

closed 2018-02-16 by don