noble/noble #756: Raspberry Pi 2B with CSR bluetooth not working (advertisement is not defined)

leiw324, 2018-02-10:
Hello SandeepMistry,

I am following the following guide to setup indoor positioning.

https://blog.truthlabs.com/beacon-tracking-with-node-js-and-raspberry-pi-794afa880318

After I run the following script (scan.js):

var noble = require('noble');

noble.startScanning();

noble.on('discover', function(peripheral) {

  var macAddress = peripheral.uuid;
  var rss = peripheral.rssi;
  var localName = advertisement.localName;
  console.log('found device: ', macAdress, ' ', localName, ' ', rss);
});
----------------------------------------------------------------------------------------------------------
I got the following error:

root@raspberrypi:~# node scan.js
/root/scan.js:9
  var localName = advertisement.localName;
                             ^

ReferenceError: advertisement is not defined
    at Noble.<anonymous> (/root/scan.js:9:19)
    at emitOne (events.js:77:13)
    at Noble.emit (events.js:169:7)
    at Noble.onDiscover (/root/node_modules/noble/lib/noble.js:174:10)
    at emitMany (events.js:108:13)
    at emit (events.js:182:7)
    at NobleBindings.onDiscover (/root/node_modules/noble/lib/hci-socket/bindings.js:174:10)
    at emitMany (events.js:108:13)
    at emit (events.js:182:7)
    at Gap.onHciLeAdvertisingReport (/root/node_modules/noble/lib/hci-socket/gap.js:256:10)
----------------------------------------------------------------------------------------------------------
This guide already old, so all of the software version didn't update, first posting what I have software version to you.

root@raspberrypi:~# lsusb
Bus 001 Device 005: ID 0a5c:21e8 Broadcom Corp. BCM20702A0 Bluetooth 4.0
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
----------------------------------------------------------------------------------------------------------
root@raspberrypi:~# npm -v noble
2.14.7
----------------------------------------------------------------------------------------------------------
root@raspberrypi:~# dpkg -l bluez
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                                  Version                         Architecture                    Description
+++-=====================================================-===============================-===============================-===============================================================================================================
ii  bluez                                                 5.43-2+rpt2+deb9u2              armhf                           Bluetooth tools and daemons
----------------------------------------------------------------------------------------------------------
root@raspberrypi:~# node -v
v4.2.1
----------------------------------------------------------------------------------------------------------

Thanks

don, 2018-02-11:
@leiw324 please edit the issue and fix the formatting so it's easier to read

The error message you are getting looks correct "ReferenceError: advertisement is not defined"

This line is the problem

     var localName = advertisement.localName;

Fix with 

     var localName = peripheral.advertisement.localName;

Note that macAdress is also mis-spelled in the console.log line. It should be macAddress.

This updated version of your scan code should work better

    var noble = require('noble');
    noble.startScanning();

    noble.on('discover', function(peripheral) {
        var macAddress = peripheral.address;
        var rssi = peripheral.rssi;
        var localName = peripheral.advertisement.localName;
        console.log('found device: ', macAddress, localName, rssi);
    });

sandeepmistry, 2018-02-11:
@don thanks, @leiw324 I suggest you contact the authors of the blog post and notify them of the issue.

closed 2018-02-11 by sandeepmistry