noble/noble #759: RP3 issue

throw-pi-at-your-face, 2018-02-11:
I don't know much about how to use this and mainly am just trying to learn more about Terminal with it but for some reason when I type "npm install noble" as root, it throws these errors and warnings at me. I'm sure I'm missing something obvious since I don't really know what I'm doing, but how fo you fix this error?

npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

npm ERR! code EBADPLATFORM
npm ERR! notsup Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm ERR! notsup Valid OS:    darwin
npm ERR! notsup Valid Arch:  any
npm ERR! notsup Actual OS:   linux
npm ERR! notsup Actual Arch: arm

don, 2018-02-11:
Make sure you're running Node.js 8.9.4 
  Follow install from https://nodejs.org/en/download/package-manager/#debian-and-ubuntu-based-linux-distributions

Make sure you have a compiler installed

    sudo apt update 
    sudo apt upgrade -y
    sudo apt install python-dev build-essential -y

Make sure you have all the [bluetooth dependencies](https://github.com/sandeepmistry/noble#linux) installed

    sudo apt-get install bluetooth bluez libbluetooth-dev libudev-dev -y

throw-pi-at-your-face, 2018-02-11:
I was in node v9.5.0, I downgraded to both v8.9.4 and v9.4.0 and still had no luck. I ran the code you suggested and had everything up-to-date. Sorry, I should have checked all that before and posted that in my original comment.

don, 2018-02-11:
It sounds like a general nodejs problem rather than a specific noble issue. I suggest you work up some exact steps to reproduce a simple problem and post to stack overflow.

closed 2018-02-11 by don