noble/noble #760: node version mismatch

srikarvijay4, 2018-02-20:
i@raspberrypi:~ $ node
> var noble = require('noble');
Error: The module '/home/pi/node_modules/noble/node_modules/bluetooth-hci-socket/build/Release/binding.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 46. This version of Node.js requires
NODE_MODULE_VERSION 57. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at Object.Module._extensions..node (module.js:672:18)
    at Module.load (module.js:556:32)
    at tryModuleLoad (module.js:499:12)
    at Function.Module._load (module.js:491:3)
    at Module.require (module.js:587:17)
    at require (internal/module.js:11:18)

sandeepmistry, 2018-02-20:
Duplicate of https://github.com/noble/noble/issues/753

closed 2018-02-20 by sandeepmistry

srikarvijay4, 2018-02-20:
Even the same problem i tried it many times will you please tell me correct versions of node npm etc to avoid this problem actually  i am working on ble beacon noble.startScanning();

noble.on('discover', function(peripheral) { 

  var macAddress = peripheral.uuid;
  var rss = peripheral.rssi;
  var localName = advertisement.localName; 
  console.log('found device: ', macAdress, ' ', localName, ' ', rss);   
)};