noble/noble #764: pending commands left unresolved on early disconnect (Linux)

pabigot, 2018-02-25:
Below is an annotated btmon trace of a situation where a Linux LE connection fails immediately after it completes.  I believe this problem underlies #465, #485, #545, and #751, and possibly others.

What seems to be happening is that, as a side effect of completing a connection the Linux kernel issues several additional HCI commands including [LE Read Remote Used Features](https://github.com/torvalds/linux/blob/v4.15/net/bluetooth/hci_event.c#L2173). Under circumstances TBD this operation can fail, which causes the connection to be shut down immediately.

This disconnection occurs after Noble has been notified of the successful connection and both library and application code has started to use it.  In the trace below the notification arrives while the exchangeMTU command initiated by [NobleBindings.onLeConnComplete](https://github.com/noble/noble/blob/v1.9.0/lib/hci-socket/bindings.js#L178) is [pending](https://github.com/noble/noble/blob/v1.9.0/lib/hci-socket/gatt.js#L199) and a user-initiated discoverServices command is still held in the Gatt instance _commandQueue.

When the disconnection is processed by NobleBindings.onDisconnComplete, the Gatt instance for the connection is [discarded](https://github.com/noble/noble/blob/v1.9.0/lib/hci-socket/bindings.js#L248) without checking its `_commandQueue` for pending operations.  As a result the callbacks that would release the application's control flow are not executed.

I believe to fix this properly requires addition of something like:

    Gatt.prototype.processDisconn = function(reason) {
      while(this._commandQueue.length) {
        this._currentCommand = this._commandQueue.shift();
        // do something here
      }
      this.removeAllListeners();
    }

along with modification of all writeCallbacks to take a parameter that indicates the connection was lost before the command could be executed, thus returning control to the application so it can recover.

    # Turn of scanning and initiate connection
    > HCI Event: Command Complete (0x0e) plen 4                    [hci0] 59.678167
          LE Set Scan Enable (0x08|0x000c) ncmd 1
            Status: Success (0x00)
    < HCI Command: LE Create Connection (0x08|0x000d) plen 25      [hci0] 60.680117
            Scan interval: 60.000 msec (0x0060)
            Scan window: 30.000 msec (0x0030)
            Filter policy: White list is not used (0x00)
            Peer address type: Random (0x01)
            Peer address: DA:CC:ED:85:8B:FD (Static)
            Own address type: Public (0x00)
            Min connection interval: 7.50 msec (0x0006)
            Max connection interval: 15.00 msec (0x000c)
            Connection latency: 0x0000
            Supervision timeout: 2000 msec (0x00c8)
            Min connection length: 2.500 msec (0x0004)
            Max connection length: 3.750 msec (0x0006)
    # Connection command acknowledged
    > HCI Event: Command Status (0x0f) plen 4                      [hci0] 60.681151
          LE Create Connection (0x08|0x000d) ncmd 1
            Status: Success (0x00)
    # Connection completes
    > HCI Event: LE Meta Event (0x3e) plen 19                      [hci0] 60.827177
          LE Connection Complete (0x01)
            Status: Success (0x00)
            Handle: 64
            Role: Master (0x00)
            Peer address type: Random (0x01)
            Peer address: DA:CC:ED:85:8B:FD (Static)
            Connection interval: 15.00 msec (0x000c)
            Connection latency: 0.00 msec (0x0000)
            Supervision timeout: 2000 msec (0x00c8)
            Master clock accuracy: 0x05
    # Feature request injected by kernel
    < HCI Command: LE Read Remote Used Fea.. (0x08|0x0016) plen 2  [hci0] 60.827312
            Handle: 64
    > HCI Event: Command Status (0x0f) plen 4                      [hci0] 60.828150
          LE Read Remote Used Features (0x08|0x0016) ncmd 1
            Status: Success (0x00)
    # Failure of kernel-injected feature request
    > HCI Event: LE Meta Event (0x3e) plen 12                      [hci0] 60.917187
          LE Read Remote Used Features (0x04)
            Status: Connection Failed to be Established (0x3e)
            Handle: 64
            Features: 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00
              LE Encryption
    # Shutdown of connection before MTU request was submitted
    > HCI Event: Disconnect Complete (0x05) plen 4                 [hci0] 60.918165
            Status: Success (0x00)
            Handle: 64
            Reason: Connection Failed to be Established (0x3e)
    # Transmission of MTU request for connection that's already gone
    < ACL Data TX: Handle 64 flags 0x00 dlen 7                     [hci0] 60.942542
          ATT: Exchange MTU Request (0x02) len 2
            Client RX MTU: 256
    > HCI Event: Number of Completed Packets (0x13) plen 5         [hci0] 61.096183
            Num handles: 1
            Handle: 64
            Count: 1
    # Nothing until application timeout shuts down pending discoverServices
    < HCI Command: Disconnect (0x01|0x0006) plen 3                 [hci0] 75.920184
            Handle: 0
            Reason: Remote User Terminated Connection (0x13)

The application code that produces this trace is something like:

    const connect = promisify(peripheral.connect.bind(peripheral));
    const discoverServices = promisify(peripheral.discoverServices.bind(peripheral, null));
    promise = stopScanning()
      .then(() => new Promise(resolve => setTimeout(resolve, 1000))); // Wait 1 s for scan to shut down
      .then(() => connect())
      .then(() => discoverServices())
      .then(() => does_not_get_here())
      .catch(() => gets_here_because_real_code_rejects_when_callbacks_not_invoked())

sandeepmistry, 2018-02-26:
@pabigot please submit a pull request with a proposal for this.

pabigot, 2018-02-26:
@sandeepmistry I don't think it's that simple.  To solve this there needs to be some common way to invoke each queued command's `callback` or `writeCallback` to indicate that the command didn't just fail, it was never invoked.

Queued commands with callbacks include:
* handleConfirmation, which callback emits `handleConfirmation`
* mtuRequest, which stores some data then emits `mtu`
* readByGroupRequest, which does something complex
* readByTypeRequest ...
* readBlobRequest
* writeRequest
* ...

For each of those somebody needs to decide what the appropriate recovery behavior should be, and in particularly how to propagate the failure through the internal event listeners to a point where it gets passed on to a user-supplied callback.

That somebody is not me.  I don't know the internals of noble anywhere near well enough to propose a solution for this.  It's sufficiently daunting that I'm planning to switch to a different solution for my own needs.

elovin, 2018-03-13:
When using multiple devices I also have a connection problem, when multiple devices try to connect to the same device. Of course only one of them can connect (BT 4.0) the others however get stuck in "connecting" and only powering on/off the device does unlock it again

elovin, 2018-03-13:
Where should I start to find the reason for this (no error is thrown on noble level)?

Jaftem, 2018-03-22:
Hi @elovin, I'm from #751. What version of BlueZ are you using? I upgraded to 5.48 and 5.49 and I have not seen this issue since.

AndyChoon, 2020-04-09:
> Hi @elovin, I'm from #751. What version of BlueZ are you using? I upgraded to 5.48 and 5.49 and I have not seen this issue since.

I do see the problem with connection to multiple devices, my bluez is 5.50