noble/noble #765: Bindings being initialized twice or more

tlongeri, 2018-02-28:
What title says.

This can be reproduced with the following

```javascript
const noble = require('noble');
noble.on('stateChange', () => {});
noble.on('stateChange', () => {}); 
x = noble.state;
```

I came across this bug when trying out [node-web-bluetooth](https://github.com/IjzerenHein/node-web-bluetooth), and using `btmon` I noticed that ATT MTU Exchange Requests were being sent 4 times when a connection was established. This was because listeners had been attached twice, and the initial MTU Exchange was called on a "chain" of 2 events.

I traced it to the lines 51-72 of `lib/noble.js`:
```javascript
  //lazy init bindings on first new listener, should be on stateChange
  this.on('newListener', function(event) {
    if (event === 'stateChange' && !this.initialized) {
      process.nextTick(function() {
        this._bindings.init();
        this.initialized = true;
      }.bind(this));
    }
  }.bind(this));

  //or lazy init bindings if someone attempts to get state first
  Object.defineProperties(this, {
    state: {
      get: function () {
        console.log("checking for 2" + this.initialized);
        if (!this.initialized) {
          debugger;
          console.log("initting bindings from 2!");
          this._bindings.init();
          this.initialized = true;
          console.log("init from 2 done!");
        }
        return this._state;
      }
    }
  });
```

So you can have initialization queued with `process.nextTick`, but `initialized` remains false. Until the initializer actually gets called you can queue any number of other initializations with `.on('stateChange',func)`, which will be later executed. You can also initialize by accessing `noble.state`, which will set `initialized` to true immediately, but will not prevent previously queued initializations to execute.

sandeepmistry, 2018-03-12:
Hi @tlongeri,

Could you please try out the changes from PR #769  and provide your feedback. Thanks.

AlCalzone, 2018-03-17:
I've been getting multiple calls of the "discover" callback since upgrading from 1.8.1 to 1.9.0. #769 fixes it for me. Please merge soon!

PR #769 (2018-03-18): Fix multiple init of bindings
By setting the initialized flag to true before calling init.

Fixes #765.
closed 2018-03-18 by sandeepmistry