noble/noble #774: Installation error in macOS High Sierra

rhumbertgz, 2018-03-22:
Hello,

I'm trying to use noble in a small project for a CoderDojo Belgium workshop next month, however, I'm having the following issue when I use `npm install noble`

```
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/bluetooth-hci-socket):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for bluetooth-hci-socket@0.5.1: wanted {"os":"linux,android,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})

+ noble@1.9.1
updated 1 package in 2.051s
```

**My setup:**

- noble: 1.9.1
- node: 8.10.0 
- npm: 5.6.0
- xcode: 9.2 (9C40b)
- macOS High: 10.13.3 (17D47)
- MacBookPro11,3

jacobrosenthal, 2018-03-22:
I dont see anything wrong there. its skipping the optional linux deps. Try
it, I think its all good to go!

rhumbertgz, 2018-03-22:
@jacobrosenthal thanks! Yes, everything is working correctly. The kids will be delighted to control a drone with a blockly-based electron app and a micro:bit

closed 2018-03-22 by rhumbertgz