noble/noble #776: Cannot discover peripherals anymore

revialim, 2018-03-23:
I picked up a small project from some time ago, in which I want to scan all bluetooth devices in my area for their uuids, but when I run my script now, it seems to not go into discover mode anymore. I might just miss something trivial... Can somebody tell me, what I'm doing wrong? Any help would be very appreciated.

Here is the script that I'm trying to run. It's basically taken from the test.js file.
```javascript
var noble = require('noble');

noble.on('stateChange', function(state) {
  if (state === 'poweredOn') {
    noble.startScanning([], true); //allow duplicates
  } else {
    noble.stopScanning();
  }
});

noble.on('discover', function(peripheral) {
  var uuid;
  if(peripheral.advertisement.serviceUuids != undefined 
  	  && peripheral.advertisement.serviceUuids != null){
  	uuid = peripheral.advertisement.serviceUuids;
  } else {
  	uuid = "undefined or other";
  }
  console.log('Found device with local name: ' + peripheral.advertisement.localName);
  console.log('advertising the following service uuid\'s: ' + uuid);
  console.log("...");
}
```

closed 2018-03-26 by revialim

Matebo, 2023-03-14:
Did you find a solution? I'm also not discovering anything.

Matebo, 2023-03-14:
Sorry, i was using @abandonware/noble. With this code the discover event is never called. Installing noble doesn't work because it cannot find module 'bluetooth-hci-socket'. I will look further into the node-ble project.

hothsolo, 2023-03-14:
Yes, I'm using the abandonware version as well.  The code I posted above works.

Matebo, 2023-03-14:
What OS are you using? I'm using Debian 11 with kernel version 5.15.5