noble/noble #783: Reset noble on HCI device reset?

Jaftem, 2018-04-10:
Hi,

I'm dealing with some BlueZ 5.49 issues where peripherals will 'permanently' stay connected ([as per this Stackoverflow discussion](https://stackoverflow.com/questions/40847659/cannot-disconnect-ble-connection)). As a workaround, I know that I can reset hci0 which will forcibly disconnect all peripherals, however Noble doesn't seem to play nice after this. Sometimes Noble will be able to receive advertisements from peripherals, but it no longer can establish a connection and hangs trying to connect.

Is there any specific/special initialization code that's executed when Noble starts? I want to be able to do:

`hciconfig hci0 reset`

or

```
hciconfig hci0 down
hciconfig hci0 up
```

And on the poweredOff or poweredOn listeners reset Noble. 

Thanks.

hunterdowns, 2018-08-24:
Yeah, this seems to be a problem in general with Noble; that is, not just Raspberry PI...I've tried both Mac and RPIs.  I have  peripheral devices that are coming in / out of range and/or being turned off and Noble doesn't seem to recover well.  I have tried disconnecting every peripheral and rescanning...no luck. On rescan, Noble will find some of the devices but usually not all. Restart the program and all is fine again...until a peripheral goes out of range or is turned off.

Did you find any solution that worked for you?

Jaftem, 2018-11-01:
@hunterdowns  We've been using crude workarounds. When the app becomes buggy, we terminate noble, reset hci0 and start the app again, which means our app is down for a few seconds. 

Below two links from the mailing list might be the issue we're having? Unsure. Are your slave devices using random addresses like ours?

- https://marc.info/?l=linux-bluetooth&m=153838396729423&w=2
- https://marc.info/?l=linux-bluetooth&m=154043205626452&w=2

RickVM, 2020-03-03:
Same situation here. Has any of you been able to properly resolve this issue?
Working with crude workarounds right now..