noble/noble #788: Docker and Noble

UsersDoProgram, 2018-04-26:
Hello, I am trying to build an application that uses an Ubuntu Docker Container with Noble but I have come across a rough patch. When trying to build and run the docker it doesn't go as expected, for one when I run docker build -t x/y, I get the errors shown (here)[https://github.com/UsersDoProgram/BlueScanner/blob/master/log2.txt], it seems that the Noble library isn't properly installed and when running the container another set of issues arise as can be seen (here)[https://github.com/UsersDoProgram/BlueScanner/blob/master/log1.txt].
I'll add the (Dockerfile)[https://github.com/UsersDoProgram/BlueScanner/blob/master/Dockerfile] and (Main.js)[https://github.com/UsersDoProgram/BlueScanner/blob/master/main.js] links so you can pick apart my mistakes. Thanks in advance! I am trying to build this on a Mac running 10.13.

sandeepmistry, 2018-05-05:
Seems ok, those only appear to be warnings.

closed 2018-05-05 by sandeepmistry

UsersDoProgram, 2018-05-09:
Yes, but when the code is run, it doesn't actually execute or wait on the changes. Execution halts once noble is required.