noble/noble #791: No longer building under node 10.0.0 / macOS 10.13.4

endquote, 2018-05-03:
Is it just me?

```
endquote@Modulo:~/Downloads$ npm install noble

> xpc-connection@0.1.4 install /Users/endquote/Downloads/node_modules/xpc-connection
> node-gyp rebuild

  CXX(target) Release/obj.target/binding/src/XpcConnection.o
../src/XpcConnection.cpp:143:89: error: too few arguments to function call, expected 2, have 1
      Local<Value> propertyValue = object->GetRealNamedProperty(propertyName->ToString());
                                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~                         ^
/Users/endquote/.node-gyp/10.0.0/include/node/v8.h:3428:3: note: 'GetRealNamedProperty' declared here
  V8_WARN_UNUSED_RESULT MaybeLocal<Value> GetRealNamedProperty(
  ^
/Users/endquote/.node-gyp/10.0.0/include/node/v8config.h:410:31: note: expanded from macro 'V8_WARN_UNUSED_RESULT'
#define V8_WARN_UNUSED_RESULT __attribute__((warn_unused_result))
                              ^
../src/XpcConnection.cpp:254:12: warning: 'MakeCallback' is deprecated [-Wdeprecated-declarations]
      Nan::MakeCallback(Nan::New<Object>(this->This), Nan::New("emit").ToLocalChecked(), 2, argv);
           ^
../../nan/nan.h:950:3: note: 'MakeCallback' has been explicitly marked deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
  ^
../../nan/nan.h:98:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
../src/XpcConnection.cpp:263:12: warning: 'MakeCallback' is deprecated [-Wdeprecated-declarations]
      Nan::MakeCallback(Nan::New<Object>(this->This), Nan::New("emit").ToLocalChecked(), 2, argv);
           ^
../../nan/nan.h:950:3: note: 'MakeCallback' has been explicitly marked deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
  ^
../../nan/nan.h:98:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
2 warnings and 1 error generated.
make: *** [Release/obj.target/binding/src/XpcConnection.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:225:12)
gyp ERR! System Darwin 17.5.0
gyp ERR! command "/usr/local/Cellar/node/10.0.0/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/endquote/Downloads/node_modules/xpc-connection
gyp ERR! node -v v10.0.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
npm WARN saveError ENOENT: no such file or directory, open '/Users/endquote/Downloads/package.json'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN enoent ENOENT: no such file or directory, open '/Users/endquote/Downloads/package.json'
npm WARN Downloads No description
npm WARN Downloads No repository field.
npm WARN Downloads No README data
npm WARN Downloads No license field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/bluetooth-hci-socket):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for bluetooth-hci-socket@0.5.1: wanted {"os":"linux,android,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/xpc-connection):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

+ noble@1.9.1
added 4 packages in 26.673s
```

JOHNEPPILLAR, 2018-05-04:
I have the same problem.

endquote, 2018-05-04:
Looks like it's an issue over in xpc-connection.
https://github.com/sandeepmistry/node-xpc-connection/issues/24

endquote, 2018-05-04:
Downgrading node seems to work.

```
brew install node@8
brew link --overwrite --force node@8
npm install noble
```

Farbod29, 2018-06-15:
thanks it works

gleeds, 2018-08-05:
This workarround fixes it without downgrading node:

https://github.com/sandeepmistry/node-xpc-connection/issues/24#issuecomment-398661672

closed 2021-03-09 by endquote