noble/noble #793: subscribe

zhaiyuzhai, 2018-05-10:
## The method Characteristic.subscribe()  does not work!!!
I have two bluetooth devices, one of which is successful and the other is not successful.
BLE adapter ::CSR8510 A10
platform:win10_64bit
1. For the first device：
`characteristic.on('data');
characteristic.subscribe(()=>{
          console.log('subscribe successfully')
});`
working successfully!!

2. For the second device:
`
characteristic.write(Buffer.from('A202A45555555555','hex'))
`
can work successfully,but:
`
characteristic.subscribe(function () {
       console.log('successfully')
});
characteristic.on(‘data’)
`
cannot get the subscribe's callback,cannot get the data.
I want to know whether the reason is noble, or adapter, or bluetooth protocol, has anyone come across it?

don, 2018-05-26:
The characteristic properties control if you can subscribe or not. To subscribe the characteristic must have notify or indicate property.

The [peripheral-explorer](https://github.com/noble/noble/blob/master/examples/peripheral-explorer.js) example will print out the peripheral details and let you see the characteristic properties.

zhaiyuzhai, 2018-05-28:
It does not work successfully!Is it because my device only supports bluetooth 4.2?

jrobeson, 2018-08-07:
No, 4.2 is fine. Do your characteristics support notify or indicate as suggested by @don?