noble/noble #796: Cannot read property 'reverse' or RangeError: Index out of range

JardarM, 2018-05-16:
Hi

I get it to works intermittently, but mostly I get the following two errors:
`RangeError: Index out of range
    at checkOffset (buffer.js:977:11)
    at Buffer.readUInt8 (buffer.js:1015:5)
    at Hci.processCmdCompleteEvent (C:\Dev\Code\Falaffel\BLE\node_modules\noble\lib\hci-socket\hci.js:566:25)
    at Hci.onSocketData (C:\Dev\Code\Falaffel\BLE\node_modules\noble\lib\hci-socket\hci.js:461:12)
    at emitOne (events.js:116:13)
    at BluetoothHciSocket.emit (events.js:211:7)
    at BluetoothHciSocket.onHciEventEndpointData (C:\Dev\Code\Falaffel\BLE\node_modules\bluetooth-hci-socket\lib\usb.js:190:10)
    at emitOne (events.js:116:13)
    at InEndpoint.emit (events.js:211:7)
    at Transfer.transferDone (C:\Dev\Code\Falaffel\BLE\node_modules\usb\usb.js:328:9)`

or
`this.address = result.toString('hex').match(/.{1,2}/g).reverse().join(':');
                                                          ^
TypeError: Cannot read property 'reverse' of null
    at Hci.processCmdCompleteEvent (C:\Dev\Code\Falaffel\BLE\node_modules\noble\lib\hci-socket\hci.js:582:59)
    at Hci.onSocketData (C:\Dev\Code\Falaffel\BLE\node_modules\noble\lib\hci-socket\hci.js:461:12)
    at emitOne (events.js:116:13)
    at BluetoothHciSocket.emit (events.js:211:7)
    at BluetoothHciSocket.onHciEventEndpointData (C:\Dev\Code\Falaffel\BLE\node_modules\bluetooth-hci-socket\lib\usb.js:190:10)
    at emitOne (events.js:116:13)
    at InEndpoint.emit (events.js:211:7)
    at Transfer.transferDone (C:\Dev\Code\Falaffel\BLE\node_modules\usb\usb.js:328:9)`

My package.json references as follows:
`"bluetooth-hci-socket": "git+https://github.com/noble/node-bluetooth-hci-socket.git",
"noble": "^1.9.1"`

My device is an Intel Wireless Bluetooth, running on Windows 10.
I reference the "bluetooth-hci-socket" directly as the npm package is not updated to support my device with VID=0x8087 and PID=0x0A2B.

Interestingly the first state change I get is 'poweredOff', I would expect this to be 'powerOn', but I might be wrong?

jrobeson, 2018-08-07:
I"m glad somebody can replicate the first state change as poweredOff. I get the same thing on windows.

You could try using my PR here: https://github.com/noble/node-bluetooth-hci-socket/pull/92 , but it requires at least node >= 6.14 and there are no available binaries, so you have to have a compiler installed locally. You can get that with the `windows-build-tools` package. You'll probably have to set the PYTHON env variable as well to point to the installed Python 2.7.

jrobeson, 2018-08-17:
@JardarM : i think i uncovered the reason for the `poweredOff` state change. It always works for me the second time i run it. I think it has to do with the device waking up from sleep too slowly. Can you verify that?

kundk-ckret, 2019-09-21:
I have the same problem :(
Worked like a charm... then i set the laptop to sleep and now i have the same issue.
Strange, that a reboot didn't fix it...

kundk-ckret, 2019-09-21:
Okay, got it to work again.

reinstalling the winUSB driver didn't help.

What worked was to reinstall the official driver for my Intel Wireless Bluetooth 8265 (yes, I'm on hci-socket 0.5.2), and then reinstalling the winUSB driver again.

Maybe the device is not correctly reset or something like that?

kundk-ckret, 2019-09-21:
Can reprodce it now on my notebook - when it goes to sleep, it is broken afterwards.

stephanpelikan, 2020-01-20:
> Okay, got it to work again.
>
> reinstalling the winUSB driver didn't help.
>
> What worked was to reinstall the official driver for my Intel Wireless Bluetooth 8265 (yes, I'm on hci-socket 0.5.2), and [...]

Thank you! I have the same issue. The errors I got where always different. Removing and reinstalling WinUSB fixes. At least for development this is OK. Maybe someone can make a hint in the Windows section of the README.md?